wled/WLED #1512: analog led-strips

x35gaming, 2020-12-17:
the chart on the readme.md file had said for analog strips, external circuitry is required, after looking through the wiki, there was no mention of external circutry.
is there some option in settings or something to allow output?

Aircoookie, 2020-12-19:
Hi! Most people use the analog support with boards that already have the necessary hardware. If you have a bare ESP board, all that you need is most likely an n channel mosfet board though, like this: https://www.amazon.com/dp/B07XJSRY6B/

There is no setting for it at this time (that will change in v0.12.0), it has to be compiled into a custom binary.

x35gaming, 2020-12-20:
is there a ws2811 compatible solution for dumb strips? i have seen ws2811 and mosfets used as a powerdac before

LeducH, 2021-01-05:
Similar question
#1034

maximecb, 2021-01-06:
@Aircoookie is there eventually going to be a prebuilt release binary for analog/PWM devices? Would really like to be able to control things like spotlights from the WLED app 👌

zer0brain, 2021-01-06:
@x35gaming 
I ordered the "h801 wifi controller" from aliexpress. these are 8266 with 5 mosfets built in and they can deal directly with 5-24V.
I have to wait until i can finally test them because of 4 weeks shipping.
Maybe someone tests those already or can get them faster.

LeducH, 2021-01-06:
> @x35gaming
> I ordered the "h801 wifi controller" from aliexpress. these are 8266 with 5 mosfets built in and they can deal directly with 5-24V.

Nice find. I might try as well. I only need the white channel but maybe with a tasmota/esphome flash this will be possible.

closed 2021-06-18 by stale[bot]