wled/WLED #1514: vscode build error (include error)

Joshude, 2020-12-17:
Hello,

I can't get the vscode enviroment running. I'm fairly new to this, flashed a few arduinos using the arduino IDE, but i'm having a hard time to get the vscode env running. I want to build a version for an analog rgbw-strip, hence the need to use vscode and build. 
The install-process described in the wiki is straight forward. install vscode, intall platformio, open the project folder and build.

**My problem is:** 
the includes don't seem to work. Sorry for the screenshot being in german, but it says: 
The file "source" cannot be opened: "Arduino.h (dependency of "NeoPixelBrightnessBus.h")...

[image]

I know, this isn't really an WLED-issue, but i hope somebody can help me here! :)

Aircoookie, 2020-12-17:
Hi!
This issue is normal before the very first compile, because only then Platform IO begins to download the required libraries.
Try compiling twice by clicking the checkmark in the bottom blue bar (the first compilation often fails due to an unrelated bug), it should build successfully and the include error should go away :)

Joshude, 2020-12-17:
Thanks! Unfortunatly that won't fix the #include error and i still can't build my code :(

Joshude, 2020-12-17:
to be on the safe side: 
- Install VSCode
- Install platformIO IDE
- Open WLED folder
- build

this should, in theory, work out of the box without any other prerequisites?

huggy-d1, 2020-12-17:
The suggestion was to rinse-repeat the last step and build a 2nd time.
The 1st build will cause PlatformIO to download and compile dependency libraries, which will arrive unfortunately out of order for a one-pass compile.
Once it finishes all the downloads, the 2nd compile will have all the files it needs as it needs them.

Joshude, 2020-12-17:
I got that, but I can build a dozent times at it will always fail. Maybe I'm missing a point in setting the vscode up or something, hence my question with the todos for setting it up :)

Aircoookie, 2020-12-17:
Sorry for the trouble. Can you send use the console output of the failing build and the changes you made to platformio.ini? :)

Joshude, 2020-12-18:
Thanks again! I've tried several things, including reinstalling VSCode, creating a project in platformio and copying the files into the project, installing git (but vscode won't recognize it) installing python.
This is the console-output. 
It doesn't matter if i use the code directly from git (without any changes) or do changes to it. 
in the platformio.h i changed only the recommended things. commented out the default boards and uncommented the d1_mini. 

```
> Executing task in folder WLED: C:\Users\Jonas\.platformio\penv\Scripts\pio.exe run <

Processing d1_mini (board: d1_mini; platform: espressif8266@2.6.2; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.7 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting wled00.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Library Manager: Installing git+https://github.com/lorol/LITTLEFS.git
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\Jonas\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Jonas\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Jonas\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Jonas\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\Jonas\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Jonas\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Jonas\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Jonas\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Jonas\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Jonas\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\Jonas\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\jonas\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\jonas\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 86:
    force=force,
  File "c:\users\jonas\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\jonas\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\jonas\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55:
    src_dir, remote_url, tag, silent
  File "c:\users\jonas\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 72:
    self.check_client()
  File "c:\users\jonas\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 138:
    "Please install Git client from https://git-scm.com/downloads"
======================================================================================================== [FAILED] Took 1.07 seconds ========================================================================================================
Environment    Status    Duration
-------------  --------  ------------
d1_mini        FAILED    00:00:01.074
================================================================================================== 1 failed, 0 succeeded in 00:00:01.074 ================================================================================================== 
The terminal process "C:\Users\Jonas\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Aircoookie, 2020-12-18:
Thanks! The error says what you need to do: `Please install Git client from https://git-scm.com/downloads`. Installing GitHub desktop might work too :) Let me know if that fixes the issue, then I will add it to the wiki!

Joshude, 2020-12-18:
Man, I feel stupid. Already installed GitHub Desktop, but that doesn't seem to be enough. After installing the "official"(?) git client it's recognized by vscode and everything works. 
So yeah, a prerequiste when starting from scratch (like i did) is indeed to install the git client from the linked source. Thanks, solved 👍

huggy-d1, 2020-12-18:
I'm glad you got it worked out!

closed 2021-06-16 by stale[bot]