wled/WLED #1515: Preset Cycle ignores the segments' sx setting. All segments have the same speed. (0.11.0)

simonjtaylor212, 2020-12-17:
**Describe the bug**
I have two presets, each with the same 5 segments configured.
In each preset, the different segments have different effect speeds. (sx)
Example:
`{"mainseg":0,"seg":[
{"id":0,"start":0,"stop":100,"grp":1,"spc":0,"on":true,"bri":255,"col":[[242,0,218],[255,255,255],[0,0,0]],"fx":50,"sx":109,"ix":132,"pal":0,"sel":true,"rev":false,"mi":false},
{"id":1,"start":100,"stop":200,"grp":1,"spc":0,"on":true,"bri":255,"col":[[242,0,218],[255,255,255],[0,0,0]],"fx":50,"sx":140,"ix":132,"pal":0,"sel":true,"rev":true,"mi":false},
{"id":2,"start":200,"stop":300,"grp":1,"spc":0,"on":true,"bri":255,"col":[[242,0,218],[255,255,255],[0,0,0]],"fx":50,"sx":183,"ix":132,"pal":0,"sel":true,"rev":false,"mi":false},
{"id":3,"start":300,"stop":400,"grp":1,"spc":0,"on":true,"bri":255,"col":[[242,0,218],[255,255,255],[0,0,0]],"fx":50,"sx":140,"ix":132,"pal":0,"sel":true,"rev":true,"mi":false},
{"id":4,"start":400,"stop":500,"grp":1,"spc":0,"on":true,"bri":255,"col":[[242,0,218],[255,255,255],[0,0,0]],"fx":50,"sx":109,"ix":132,"pal":0,"sel":true,"rev":false,"mi":false},
{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}
]}`

If I call these presets manually, I see, as expected, the different segments with different speeds.
If I use "Preset Cycle" to cycle through these presets, all segments have the same speed.

**To Reproduce**
Use Preset Cycle to cycle through two presets where sx is different for each segment.

**Expected behavior**
I expect the different segments to operate at different speeds, like they do when I manually recall the presets.

**WLED version**
 - Board: D1 Mini v3
 - Version 0.11.00
 - Format Standard Binary

**Additional context**
Thanks for your hard work on this project!

fatness-uk, 2020-12-18:
I have had the exact same problem but I thought I must be doing something wrong.  I have 3 segments set to light up at different times to each other and the sequence is saved to a preset and the preset works when manually selected but when the preset is part aof a cycle then the timing for all 3 segments are the same.

simonjtaylor212, 2020-12-24:
This was fixed in 0.11.1

closed 2020-12-24 by simonjtaylor212