wled/WLED #1516: Saving MQTT password with special characters causes sync settings page to be inaccessible

thedeany, 2020-12-17:
**Describe the bug**
After adding an MQTT username and password, the sync settings page is inaccessible. I can inspect the network request and see the HTML/JS gets mangled around setting the `d.Sf.MQPASS.value` value. My MQTT password _does_ contain `%` and `!` (the rest is hexadecimal), and I'm assuming one or both of those characters is causing the issue.

**To Reproduce**
Add an MQTT password with `%` and (possibly or) `!` in it and save.

**Expected behavior**
To be able to save a strong password with special characters.

**WLED version**
 - Board: NodeMCU
 - Version: 0.11.0
 - Format: binary

Thank you for your help!

thedeany, 2020-12-17:
I don't know much about C++, but it seems the issue is isolated to the use of `%` in the password. I was able to send a POST request to http://wled_IP/settings/sync with form data `MQPASS: new_pass` and I could access the page once again (though other settings on that page seemed to have been blown out; not sure if that's related to my POST request or if the errant password issue caused it.)

After accessing the page, I set a password with a `!` again and some other special characters, but I can still access the page after saving with that value. So I'm good to go here, but this issue should probably stand so `%` can be handled properly in that (and I'm assuming all other) input fields.

Aircoookie, 2020-12-17:
Hi!
Thank you for the extra testing! Indeed I could confirm that a `%` character in any textbox (except the wifi and AP password) will lead to the issue for the respective settings page.

For passwords, it shouldn't be an issue, since those are not re-transmitted to the settings HTML, instead only a string of `*` with the length of the password is transmitted to reduce the chance of MITM attacks or similar. An exception is the MQTT password, which is apparently transmitted in clear. Not good at all and definitely a security issue.

Will make that use the same logic as the other passwords, that will also take care of your specific case, but not of the base problem that no text field may contain a single `%`. Will think about how to fix that (if you are wondering, [this](https://github.com/Aircoookie/WLED/issues/1510#issuecomment-747359950) is likely the cause)

commit 7483d3b2 (2021-06-30): Fixed settings page broken by using "%" in input fields (fixes #1516 )
closed 2021-06-30 by Aircoookie