wled/WLED #1517: VSCode env setup for Mini D1 ESP32

ledRookie, 2020-12-17:
Hi,
I just got my QuinLED-Dig-Uno (ESP32 version) board (still awaiting the LED strip to come in the mail) and successfully flashed the latest WLED (0.11.0) version (WLED_0.11.0_ESP32_ledpin16.bin) on it; I was able to configure my network setup and connect with the app on my mobile phone.

Next, I installed VSCode and PlatformIO, compiled WLED and flashed my board. The issue I have is that with my own complied code - the app times out accessing the ESP32 board; While WLED-AP works as expected and my network credentials are entered - the ESP32 WiFi never connects to my WiFi network (checked all clients on my network). Flashing back the pre-compiled binary gets everything to work as expected;

This is the ENV I used in platformio.ini -

[env:wemos_d1_mini32]
platform = espressif32
board = wemos_d1_mini32
build_unflags = {common.build_unflags} build_flags = {common.build_flags_esp32} -D LEDPIN=16
lib_ignore =
ESPAsyncTCP
ESPAsyncUD

I think that my env configuration is incorrect but not sure what it should be set to (I tied dozens of flags/options without success).
If the [env] setting used to generate WLED_0.11.0_ESP32_ledpin16.bin can be shared - it would be of some great help.

Thank you in advance.

Aircoookie, 2020-12-17:
Hi!
Your environment seems ok - apart from formatting which I assume is due to pasting it here.

In any case, here is the environment for GPIO16 - it is also part of the platformio.ini:

```ini
[env:custom32_LEDPIN_16]
board = esp32dev
platform = espressif32@2.0
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp32} -D LEDPIN=16 -D RLYPIN=19
lib_ignore =
  ESPAsyncTCP
  ESPAsyncUDP
```

Hope you can get it to work :)

ledRookie, 2020-12-17:
Aircoookie,
Thank you for your quick reply to this.
Unfortunately, this did not fix the issue. I now recall that I tried this env ([env:custom32_LEDPIN_16]) setting before with the same result.

However, after continuing debugging this here is what I noticed:
**With the pre-compiled WLED_0.11.0_ESP32_ledpin16.bin:**
- I keep the default of: AP opens: No connection after boot
- After entering and saving my WiFi credentials - WLED-AP network is not detected by my phone (as expected).
- My d1 mini board is detected on my WiFi network

**With my own compiled custom32_LEDPIN_16.bin bin:** 
- I keep the default of: AP opens: No connection after boot
- After entering and saving my WiFi credentials (CLIENT_SSID is pre-set in my_config.h) - WLED-AP network is **still** on and is detected by my phone (unexpected).
- My d1 mini board is not detected on my WiFi network

Which leads me to believe that my WiFi credentials/settings are not saved in my own compiled WLED code and so - when I re-boot my d1 mini board - it won't appear on my WiFi network...

Is there a way for me to check/confirm that my WiFi credentials are properly saved?

ledRookie, 2020-12-18:
I did a quick experiment with WLED 0.10.2 using env:custom32_LEDPIN_16 and am able to compile, flash and connect to my board successfully as expected.

This seems to suggest that there is somewhat a difference behavior when locally compiling 0.11.0.  
Could it be that the posted custom32_LEDPIN_16.bin was generated with some code changes that were not checked in and released with 0.11.0?

Thanks!

ledRookie, 2020-12-21:
I got it all working now.
It turns out my 0.11.0 environment got corrupted and after creating a fresh/clean environment, pointing to env:custom32_LEDPIN_16 everything works as expected!

Very sorry for wasting your time on this.

Aircoookie,
I want to take a moment and really thank you for putting this project together!
Got it all working and the effects were amazing! 
Much appreciated!!

closed 2020-12-21 by ledRookie