wled/WLED #1521: Sync not working

superbond05, 2020-12-18:
I noticed that my lights are not syncing anymore. First I could not find one of my controller on the computer or iphone app. searched for it for two days it would not come back on line. Was scared to reset as I might loose all me presets. I pressed the reset button on the controller and all is good. I did not loose my presets but I used this controller to control all my effects for the rest of the house. 
 I tried to unsync and resync a dozen times with no success. How do I get them to Sync with this controller again?

**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

WoodyLetsCode, 2020-12-18:
Did you also restart all your other controllers?
And btw: You can also restart them in the wled web page. Just click on info.

superbond05, 2020-12-18:
no But I will

superbond05, 2020-12-18:
Refresh or reboot?

WoodyLetsCode, 2020-12-18:
Reboot

superbond05, 2020-12-18:
K

superbond05, 2020-12-18:
Nice 
Seems to be working! Thank you

closed 2020-12-18 by superbond05