wled/WLED #1522: Shelly RGBW2 Firmware

djashjones, 2020-12-18:
Is the binary available for the shelly or do I have to compile it myself?

WoodyLetsCode, 2020-12-18:
Here you can find all available binary files: https://github.com/Aircoookie/WLED/releases/latest

djashjones, 2020-12-18:
"For "analog" LED use only! Runs on 12-24VDC. One button and one input. Pins: R=12, G=15, B=14, W=4. Finished, commercial product that can be flashed. Firmware fork is here".

The "Firmware Fork is here" is the bit that has me confused

huggy-d1, 2020-12-18:
Where did you find the "firmware fork is here" reference?

djashjones, 2020-12-18:
"Compatible Hardware" in the WLED Wiki section.

huggy-d1, 2020-12-18:
When you followed the link, did you find the firmware?  @srg74 has not updated the release builds on github since 2019 for that github wled fork, but the released firmware is found at this link: https://github.com/srg74/WLED/releases
If you want to compile the latest wled version, you can of course download the source from the same site and do that.

djashjones, 2020-12-18:
No firmware but the source code which as you said is outdated

huggy-d1, 2020-12-18:
https://github.com/srg74/WLED/releases/tag/v0.9.0-b1 is latest "released" firmware.  In all seriousness, it is probably sufficient to do exactly what you want with a Shelly to control RGBW LED strips.

The source code has apparently been kept up to date.  @srg74 merged the latest source code just yesterday.

djashjones, 2020-12-19:
I can not see a binary for the shelly. So I'm going to try to see if I can compile it

srg74, 2020-12-19:
> I can not see a binary for the shelly. So I'm going to try to see if I can compile it

I guess you have to click on word here in my post and you will see binaries you looking for.

djashjones, 2020-12-19:
Fantastic, That was not there the last time I looked

closed 2020-12-20 by djashjones

al3x1337, 2021-09-17:
> That is only a release tag. Code is up to date and all binaries is marked accordingly. So there is shellyRGBW2 latest [here](https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/Shelly_RGBW2)

The link does not work anymore and I can't find the binary when searching for Shelly or RGBW2 in your profile. Is WLED still available for the RGBW2 anymore in any way?

Aircoookie, 2022-03-02:
@SummerSeaSun thanks, I removed the broken links.
A separate binary/firmware is no longer required as the pins can be confugured in LED settings (LED type: `PWM RGBW`)

ikavior, 2023-03-01:
Maybe someone knows, can we use button pin (gpio13) as a data for addressable LEDs? Like WS28xx?

blazoncek, 2023-03-01:
Yes but with caveats. TL:DR not good idea.