wled/WLED #1523: MOD: Add 433 MHz transmitter feature for control RC socket

tlgWTF, 2020-12-18:
**Idea**
You have a *low-power* USB power supply and a *powerful* LED power supply that only needs power when the LEDs are on.

[**Fork link**](https://github.com/GlitchPunkWTF/WLED/) | [**Commit link**](https://github.com/GlitchPunkWTF/WLED/commit/f62bba13fee3335e26d32e8e716973d996bc782a)

**Basic implementation**
You have a relay that breaks the wire to the LED power supply. But you need to cut the live wire with a relay, which can be dangerous due to the poor quality of the relay module.

[image]

**My implementation**
You have a stock 433MHz RC power socket that breaks power for LED power supply.
**Advantages:** you don't need to cut the power cable
**Disadvantages:** use of externally vulnerable radio protocols

> All configs are made with defines and written as new build config. The assembly uses the [rc-switch](https://github.com/sui77/rc-switch) library as a dependency.

[image]

>Disclaimer! I`m noob in PlatformIO

Thanks everyone for the WLED project!

killthekid, 2020-12-21:
Hey, did you add this kind of  433mhz tx to the esp? 
[image]

tlgWTF, 2020-12-21:
Yes, I used the RX module and the sample sketch for it to get the keycodes and define them as build_flags (RCON/RCOFF), so now it only uses the TX module and transmits the command at the same time as the relay module in the original program. Works well at 3.3 volts over a short distance (about 0.5 meters / 20 inches) without an antenna. TX connected to GPIO15 (D8) (RCPIN in build_flags).