wled/WLED #1525: DMX out: mapping from DMX channels to functions are not saved after reboot

mherweg, 2020-12-19:
**Describe the bug**
mapping from DMX channels to functions(shutter, red,green,blue) are not saved after reboot.
Instead they are reset to "Set to 0"

**To Reproduce**
In the DMX menu map functions to some DMX channels.
click "Save"
Power off and on again.

**Expected behavior**
mapping from DMX channels to functions(shutter, red,green,blue) do survive reboot and power cycles.

**WLED version**
 - Board: Wemos D1 mini
 - Version: 0.11.0
 - Format: self-compiled

**Additional context**
Thank you for WLED. I love it. I love all the features and the TPM2.net support

Aircoookie, 2020-12-19:
Hi, thank you for the report and I'm happy that you like WLED!
Can you send me the output of your `http://<WLED-IP>/cfg.json` file? Don't worry, it does not include your wifi password.

mherweg, 2020-12-19:
Thank you for the very quick reply! With @JvPeek's help I found out that this issue does happen with WLED 0.10.2.

mherweg, 2020-12-19:
{
  "rev": [
    1,
    0
  ],
  "vid": 2012160,
  "id": {
    "mdns": "wled-ab0e56",
    "name": "WLED",
    "inv": "Light"
  },
  "nw": {
    "ins": [
      {
        "ssid": "foobar",
        "pskl": 8,
        "ip": [
          0,
          0,
          0,
          0
        ],
        "gw": [
          0,
          0,
          0,
          0
        ],
        "sn": [
          255,
          255,
          255,
          0
        ]
      }
    ]
  },
  "ap": {
    "ssid": "",
    "pskl": 8,
    "chan": 1,
    "behav": 0,
    "ip": [
      4,
      3,
      2,
      1
    ]
  },
  "wifi": {
    "sleep": true,
    "phy": 1
  },
  "hw": {
    "led": {
      "total": 30,
      "maxpwr": 850,
      "ledma": 55,
      "rev": false,
      "rgbwm": 3,
      "ins": [
        {
          "en": true,
          "start": 0,
          "len": 30,
          "pin": [
            2
          ],
          "order": 0,
          "rev": false,
          "skip": 0,
          "type": 22
        }
      ]
    },
    "btn": {
      "ins": [
        {
          "type": 2,
          "pin": [
            0
          ],
          "macros": [
            0,
            0,
            0
          ]
        }
      ]
    },
    "ir": {
      "pin": 4,
      "type": 0
    },
    "relay": {
      "pin": 12,
      "rev": false
    },
    "status": {
      "pin": -1
    }
  },
  "light": {
    "scale-bri": 100,
    "pal-mode": 0,
    "gc": {
      "bri": 1,
      "col": 2.8
    },
    "tr": {
      "mode": true,
      "dur": 7,
      "pal": 0
    },
    "nl": {
      "mode": 1,
      "dur": 60,
      "tbri": 0,
      "macro": 0
    }
  },
  "def": {
    "ps": 0,
    "on": true,
    "bri": 128
  },
  "if": {
    "sync": {
      "port0": 21324,
      "port1": 65506,
      "recv": {
        "bri": true,
        "col": true,
        "fx": true
      },
      "send": {
        "dir": false,
        "btn": false,
        "va": false,
        "hue": true,
        "macro": false,
        "twice": false
      }
    },
    "live": {
      "en": true,
      "port": 5568,
      "mc": false,
      "dmx": {
        "uni": 1,
        "seqskip": false,
        "addr": 1,
        "mode": 4
      },
      "timeout": 25,
      "maxbri": false,
      "no-gc": true,
      "offset": 0
    },
    "va": {
      "alexa": false,
      "macros": [
        0,
        0
      ]
    },
    "blynk": {
      "token": ""
    },
    "mqtt": {
      "en": false,
      "broker": "",
      "port": 1883,
      "user": "",
      "pskl": 0,
      "cid": "WLED-ab0e56",
      "topics": {
        "device": "wled/ab0e56",
        "group": "wled/all"
      }
    },
    "hue": {
      "en": false,
      "id": 1,
      "iv": 25,
      "recv": {
        "on": true,
        "bri": true,
        "col": true
      },
      "ip": [
        192,
        168,
        8,
        0
      ]
    },
    "ntp": {
      "en": false,
      "host": "0.wled.pool.ntp.org",
      "tz": 0,
      "offset": 0,
      "ampm": false
    }
  },
  "ol": {
    "clock": 0,
    "cntdwn": false,
    "min": 0,
    "max": 29,
    "o12pix": 0,
    "o5m": false,
    "osec": false
  },
  "timers": {
    "cntdwn": {
      "goal": [
        20,
        1,
        1,
        0,
        0,
        0
      ],
      "macro": 0
    },
    "ins": []
  },
  "ota": {
    "lock": false,
    "lock-wifi": false,
    "pskl": 7,
    "aota": true
  },
  "dmx": {
    "chan": 6,
    "gap": 6,
    "start": 1,
    "start-led": 0,
    "fixmap": [
      6,
      1,
      2,
      3,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0
    ]
  },
  "um": {}
}

Aircoookie, 2020-12-19:
Thank you! It does get saved in the file, so that doesn't seem to be the issue. Then it would also make sense if it doesn't work in 0.10.2...

mherweg, 2020-12-19:
I'm so sorry! there was the word "not" missing in my last comment ;-)  I corrected my statement above. sorry for the confusion!

mherweg, 2020-12-19:
yes it stays saved in the json file after reboot , but it is not used in the code or in the GUI after the reboot

mherweg, 2020-12-19:
I think this line:

JsonArray dmx_fixmap = dmx.**createNestedArray**("fixmap");

in cfg.cpp   is wrong because it's the only time I see it  inside   deserializeConfig()

in all other cases createNestedArray is only used inside  serializeConfig()

mherweg, 2020-12-19:
please try this diff

https://pastebin.com/u3rYMwJs

I think it works for me. I'll do some more testing.

windows10luke, 2021-05-17:
I am having this exact same issue in the latest build of 0.12.0. It stopped working after the 0.10.2 release.

windows10luke, 2021-09-20:
This issue still persists in the latest build of WLED.

PR #2328 (2021-11-09): Fix a bug which prevents DMX Output channel mappings from loading correctly
This fixes #1525.

As far as I can tell, this bug has existed since the introduction of DMX output. It was likely hard to manifest because the common mapping
```
channel 1 -> red
channel 2 -> green
channel 3 -> blue
```
just happens to work. Any other mapping would not load properly.
closed 2021-11-09 by Aircoookie