wled/WLED #1527: 11.1 cannot save preset transition timings

AlroyLED, 2020-12-19:
Just updated to 11.1 tried iOS and pc versions but cannot save the time per transition. Is there a save button missing, tried enter key but every time I leave the page and return it’s reverted to 1.2seconds.

just a warning for anyone who is using for Christmas lights.  Still loving WLED though - keep up the good work.

AlroyLED, 2020-12-19:
UPDATE - I can save only if I disable the preset cycle first.   Minor bug now.  Could do with these being greyed out whilst it’s ticked to indicate non editable until preset is disabled first.

feel free to close if not required?

Aircoookie, 2020-12-19:
Hi!
Yes, indeed this is a little annoyance. The preset cycle duration (and start and end presets) will only get sent if the preset cycle is enabled or disabled with the checkbox. Other commands or refreshing the page will reset it to the old values. I'm taking this as a suggestion to improve the behavior :)

blazoncek, 2021-10-05:
Preset cycle has been replaced by playlists in 0.13.
Closing.

closed 2021-10-05 by blazoncek