wled/WLED #1530: Need segments with arbitrary LED numbers.

booroondookZ, 2020-12-19:
It is need to have segments with arbitrary LED numbers, not consecutive numbers only.
Sometimes the physical placement of the LEDs cannot be match their placement on the cable.
Suppose we have a curtain with attached LEDs.
We want to make the segments as horizontal rows.
Segment #0 should include LEDs 0, 1, 12-15, 26-29, 40 and 41.
Segment #1 should include LEDs 2-4, 9-11, 16-18, 23-25, 30-32 and 37-39.
Segment #2 should include LEDs 5-8, 19-22 and 33-36.
The current settings do not allow including arbitrary LED numbers in the segments, but only allow sequences of strictly following numbers.
[image]

Aircoookie, 2020-12-20:
Hi!
This is a really good enhancement - however I'm not yet sure how it could be added. This goes more into a direction of a custom mapping (that would allow you to define the top "segment" as LEDs 0-12, even though it is 0, 1, 12-15, 26-29, 40, 41 in hardware).
Segments themselves will likely stay the way they are - as a continuous area of LEDs (unless I get a really good idea!)
Otherwise you would also get the problem that you don't know in what order the segment should be addressed (0, 13, 14, 27, 28, 41, 1, 12, 15, 26, 29, 40), so horizontal lines or maybe rather (14, 13, 0, 1, 12, 15, 26, 29, 40, 41, 28, 27), wrapping around from the middle.
In case you would like to use this custom mapping, there is an array in `FX_fcn.cpp` where you can map a virtual LED (how it is addressed in WLED effects and segments) to a physical LED :)

booroondookZ, 2020-12-20:
I think it is possible to create a table with the correspondence between the physical and logical LED numbers. And then in all other settings, operate only with logical numbers.
Also, this table should have a "Reset" button, when pressed, all logical numbers become equal to physical ones.

blazoncek, 2023-05-27:
WLED 0.14+ supports matrices where OP request can be done by simply using segments.
If you need 1D representation then you can use [*ledmaps*](https://kno.wled.ge/advanced/mapping/).

closed 2023-05-27 by blazoncek