wled/WLED #1537: When using static IP set the WLED does not resolve dns queries for MQTT server

daromer2, 2020-12-20:
**Describe the bug**
If i set the IP to static it wont resolve the mqtt server that i have running with dns name

**To Reproduce**
1. Set wifi to use static ip
2. Set MQTT server to a dns that of course resolves to your mqtt server. Examle mqtt.test.com if thats what you have

**Expected behavior**
It should resolve the dns to what you have and connect. But it looks like it doesnt. Missing DNS config when using static ip?
I use static ip because the api frm HA and other is referring to it like that. 

**WLED version**
 - Board: [e.g. Wemos D1 mini
 - Version 0.11.1
 - Format binary release. ESP8266

**Additional context**
Nothing else to say. I think the issue is pretty self explanatory :) 

Thank you for your help!

Legsmaniac, 2020-12-20:
Hmmmmm.  Strange.  I'm using MQTT on a Wemos D1 Mini with Static IP no issues.
Something amiss here. 🤔

daromer2, 2020-12-20:
@Legsmaniac 
Do you use IP for your MQTT broker or dns?  
When you set static IP on t he wled what dns server do it use?

Legsmaniac, 2020-12-20:
My MQTT broker (Mosquito) is running on a spare PC with a static IP used as a CCTV and FTP server.

Aircoookie, 2020-12-20:
@daromer2 is the Static gateway set to the IP address of your router?

daromer2, 2020-12-20:
Aircookie yes it is. I guess the issue is that you rely on the GW being the DNS. In my case that's not correct. When using static the DNS should either listen to DHCP info or be set static and not via GW. 

My DNS server sits on different ip/network even.

Aircoookie, 2020-12-22:
Hi, I just double checked this and indeed I do not use the GW as DNS, but hardcoded `8.8.8.8`. Is that accessible in your network setup? If not, I might need to add fields to set a custom DNS IP. You can change it in line 405 of `wled.cpp`.

daromer2, 2020-12-22:
8.8.8.8 is accessible in my setup but i have split dns where all my internal dns entries are of course only reachable from my internal networks :)  Surprised that im the only one so far reporting it but many home users either dont use dns for services like that or they expose their internal lookup externally :)

I leave it for now. I dont need to and dont want to custom compile for now. Easier to temporarily expose my mqtt dns externally as well. Its not really a security issue as of today.  

Custom field for dns would be good when setting static ip but could be pre-filled in with google dns (8.8.8.8) i guess :)

huggy-d1, 2020-12-23:
@daromer2 You could configure your DNS server to route 8.8.8.8 to your local DNS server rather than exposing any IoT device to the internet for anything.

daromer2, 2020-12-23:
@huggy-d1 you mean faking that 8.8.8.8 is my local dns?  I dont see that as a good way of doing it though :) The best way is to make sure you can set a proper internal dns from the start :)

huggy-d1, 2020-12-23:
@Aircoookie offered you a way to do that, but you refused that option.  I gave you a hack/work-around and you refused that option.  Happily though you refused, it still does not get it done 👍 

If you search YouTube for a recent video by Quindor, you will see exactly how to compile WLED from scratch.  I recommend that option and within an hour (probably less since you can speed it up and skip lots of it), your DNS entry will be proper "from the start".  Up to you of course.

daromer2, 2020-12-23:
Huggy not sure what you search for? Im not refusing anything but they are workarounds.  Im more than happy for work arounds and Aircookies is a decent one for sure. Especially since its determined to be by design.  If it was an issue that prevented me from running i would gladly compile it and change the code but it isnt. I can run it with IP and i can also add the entry to the external side of my dns servers.  It doesnt take 1 hour to compile it though since i do that every other day with other software :)
If i would start to recompile i would need to do that on every release and currently im more than happy staying on the releases. 

Hacking the network and pretending to have google DNS servers internally seem a bit over the top for this case. Also since the 3rd DNS in the priority list is Google DNS makes it a bit awkward doing that way to be honest. 

I will be waiting for the enhancement or even submit a pull request myself for the change needed :) 

No hard feelings but do note that its not an issue preventing me from running it. Its just a design-flaw :)

blazoncek, 2023-05-27:
If you choose to forbid/disable DNS forwarder on your GW (usually your router) or you have a complex set-up where your DNS server is external to GW then I would recommend to stop using static IP address and start using DHCP with permanent reservations. In such case you'll be able to provide WLED customised GW and DNS server addresses.

bezigebever, 2023-12-12:
I had issues with NTP in WLED, now when debugging, it seems WLED uses 1.1.1.1 as DNS server. I have hardcoded IP and can't configure the DNS server anywhere.
I don't want to open my IoT network to the outside public DNS, as this poses security risks.
Why can't the DNS be made configurable via the WLED GUI, like the other network parameters ?