wled/WLED #1538: SK6812 RGBW Not working

LastNightHoe, 2020-12-20:
**Describe the bug**
When i Activat the RGBW with my SK6812. The LED shows fancy colors i dont want.

**Expected behavior**
on the 0.10.2 it works fine. 

**WLED version**
 - Board: ESP8266
 - Version v. 0.11.1
 - Format [e.g. Binary, self-compiled]

Thank you for your help!

Aircoookie, 2020-12-20:
Hi, can you check the RGBW is still enabled in LED settings? What sort of "fancy colors" does it show? It it blinking or a solid, repeating pattern?

LastNightHoe, 2020-12-20:
i tried RGBW on with Dual, acurate, RGB GBR all. I made a video i hope it helps u. :)

https://user-images.githubusercontent.com/76401938/102722986-3ea79680-4305-11eb-82b2-94fd8a78850a.mp4

mike2nl, 2020-12-21:
@LastNightHoe @Aircoookie 
i had the same adter compiling and flashing my sonoff SV. I played a bit with the LED settings and i got it right. I used pio 3 (RX) for the data channel. That has to do with some extra sensors for tests.

I use these LED's: 32 Bit SK6812 Rgbw Led Stick Bars to simulate a fire place for christmas.
https://nl.aliexpress.com/item/32822732082.html?spm=a2g0s.9042311.0.0.27424c4d83Bgmv

My LED settings about thgese LED's are:
[image]

I hope that helps.

Aircoookie, 2020-12-21:
@LastNightHoe at the end of your video, when you disabled RGBW, it seemed to show the correct colors. I think the strip you have is not RGBW, but plain RGB...

LastNightHoe, 2020-12-21:
On Version 0.10.2 it works great. I flash this version on the ESP so it works like it should. (: 
But it’s Strange it want work on 0.11.0 and 0.11.0