wled/WLED #1544: Alexa group assignment only adds the same WLED light no matter which one is chosen to add

StefanRu1, 2020-12-20:
**Describe the bug**
I updated today from 10.2 to 11.1.
The Alexa devices did not work. So i have done a new search and the devices where discovered again with the same names.
I think there was a fix for Alexa discovery so internal descriptor or something has changed.
So i deleted the old lights and it seemed to be good. 
I can switch lights on/off by their name.
Then i wanted to add the lights again to my groups. Like Büro.
But always when i add a WLED light, in my case "Bürolicht", "Küchenlicht" gets added.
I could not believe this and deleted all lights again. Did a rediscovery and tried again.
No matter which room i use, i also created new ones, no matter which WLED i want to add, it is always "Küchenlicht" which is added.
I think something is wrong with the internal description of the lights?

**To Reproduce**
Have at least too WLED lights, create a new Group in Alexa. Add both lights. 
Result: only one light is added.

This will be the same for all WLED lights. Only this one light will be added.

**Expected behavior**
It should be possible to add a WLED light to a group and this light should be the added one. Not another one which was not choosen.

**WLED version**
 - Board:  Wemos D1 mini
 - Version 11.1
 - Format Binary

Thank you for your help!

Aircoookie, 2020-12-21:
Hi and thank you for the report!
Oh no - that gives me a deja vu - we already had this problem and the fix was giving all devices unique ids (which doesn't refer to the `uniqueid` field of the light, but to the internal light id in the hue api). In the progress of fixing the discovery in Espalexa 2.5.0 / WLED 0.11.1, this was reverted to the previous behavior of just having ID 1 for every light. Normally this should work fine, but the Alexa implementation appears incapable of dealing with multiple "Hue bridges" on the network when it comes to its Groups feature - which I did not test. I hope that by reverting to the previous ID system, but keeping the 12-character `uniqueid`, we can solve both the discovery and grouping issue.

StefanRu1, 2020-12-21:
Ok, thanks for the fast answer and the upcoming fix!

I can say the discovery worked perfect :-) 
Before the 11.1 had problems with the discovery and had often to restart my whole routers and do some tricks to get the lights discovered. Now it worked without problems.

Thanks,
Stefan

MikeOfGit, 2021-01-04:
Hi guys, thanks for bringing this topic up. I like to add an info this issue (it happens to me also > 2 WLED devices in my house for now):
As the topic opener described, the name of the first ever detected WLED will be used after grouping. Applying a command (spoken or via App) to that group will affect only the first WLED device. Addressing the new device by it's name will handle it correctly.
Cheers
Mike

StefanRu1, 2021-01-04:
Hi Mike,

yes if i ask Alexa for a specific light name it works. But in groups only the first WLED is switched.
So essentially if you have more than one WLED light you can not use group assignments to switch your lights.

So for me this means "Alexa switch off living room", will not switch off WLEDs in my living room, but in the kitchen :-(

But thanks for the clarification, the behavior is a little bit complex to describe :-)

Thanks,
Stefan

TechnoHoschi, 2021-02-11:
hi,

any news about this?

thanks
micha

geforcefan, 2021-02-15:
@TechnoHoschi I also couldn´t wait for a fix. Gonna commit a fix on a forked repo, its now working for me.... Maybe I could PR my fix @Aircoookie?

nemy77, 2021-02-23:
I'm having this same issue and its really screwing up my project to not be able to group multiple strips together with 1 command.  I can fix it by changing the unique ID in the API?  Can someone tell me how to do that?

geforcefan, 2021-02-23:
@nemy77 I posted a solution right above

nemy77, 2021-02-23:
> posted

maybe i'm having a duh moment here, but i cant figure out how to change the unique ID, are you doing that in the alexa app?

geforcefan, 2021-02-24:
@nemy77 its a bug in the wled software, I provided a bug fix in my own fork, you need to compile the binaries by yourself according to the wiki post about "compiling"

nemy77, 2021-02-24:
> @nemy77 its a bug in the wled software, I provided a bug fix in my own fork, you need to compile the binaries by yourself according to the wiki post about "compiling"

Thank you!! that was exactly the directional help I needed, I now have them working in groups properly

Aircoookie, 2021-02-24:
Sorry for the delay in getting to this! I will publish an updated version of my Espalexa library and the fix will subsequently be added to WLED 0.12.0 :)

StefanRu1, 2021-04-09:
Hi, 
i updated all my WLED devices and it works great!
Thank you very much for the fix.

From my side we can close the issue.

Thanks and best regards,
Stefan

closed 2021-04-16 by Aircoookie