wled/WLED #1546: Visual feedback LEDs

lanman1, 2020-12-21:
**Describe the solution you'd like**
I would like to be able to dedicate X numbers of pixels at the beginning of a string to use as visual indicators of which preset is playing.  I envision using this to visually see which preset is running locally at the controller so I can use the button to scroll through the presets and have the corresponding pixel illuminate for the sequence in use.  For example, preset 3 would illuminate pixel number 3 while the sequence is running.  I would think this would be easiest to implement by just having an interface to pick how many LEDs to use for this (a special segment maybe) and then have the regular leds just begin after these, all on the same physical string.

**Describe alternatives you've considered**
I considered using the OLED display for this but didn't have success and making a clean looking install would be more difficult.  I also wonder if a mini segment of say 5 leds would be a good enough indicator of what the string was doing, I also worry that in an automotive application, that this mini string flashing would be a distraction at night.

Aircoookie, 2020-12-21:
Hi!
It should be possible to do something like this in v0.11.1 natively, however it _does_ require manual additions to the presets.
Suppose you save a preset, it will contain a JSON api command similar to this:
```json
{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":8,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,255,255],[0,0,255],[0,0,0]],"fx":41,"sx":253,"ix":255,"pal":0,"sel":false,"rev":false,"mi":false},{"id":1,"start":8,"stop":53,"grp":1,"spc":0,"on":true,"bri":255,"col":[[102,153,0],[0,0,0],[0,0,0]],"fx":101,"sx":128,"ix":128,"pal":50,"sel":false,"rev":false,"mi":false},{"id":2,"start":53,"stop":183,"grp":1,"spc":0,"on":true,"bri":255,"col":[[0,204,51],[0,0,0],[0,0,0]],"fx":65,"sx":199,"ix":0,"pal":54,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"
```

Now you can set the first segment 0 to however many LEDs you want to use for the indicator. After saving the preset, modify the json command by adding something like `"i":[0,[255,0,0]],` (this would turn the first LED to red) to the beginning of the segment object. This lets you set the LED # you want to turn on (probably you want to set LED `preset number -1` here) and the color to set it to. You can insert this directly after the `"seg":[{` part in the JSON command and end up with something like this:
```json
{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"i":[0,[255,0,0]],"id":0,"start":0,"stop":8,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,255,255],[0,0,255],[0,0,0]],"fx":41,"sx":253,"ix":255,"pal":0,"sel":false,"rev":false,"mi":false},{"id":1,"start":8,"stop":53,"grp":1,"spc":0,"on":true,"bri":255,"col":[[102,153,0],[0,0,0],[0,0,0]],"fx":101,"sx":128,"ix":128,"pal":50,"sel":false,"rev":false,"mi":false},{"id":2,"start":53,"stop":183,"grp":1,"spc":0,"on":true,"bri":255,"col":[[0,204,51],[0,0,0],[0,0,0]],"fx":65,"sx":199,"ix":0,"pal":54,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
```
Note that this will disable any effect on Segment 0 and set all LEDs but the one explicitely turned on to black.

See https://github.com/Aircoookie/WLED/wiki/JSON-API#per-segment-individual-led-control for more info and let me know if this works for your use case :)

blazoncek, 2022-07-25:
You can use cheap 4 line OLED display and rotary encoder using 2 usermods with the same name.
Closing

closed 2022-07-25 by blazoncek