wled/WLED #1547: Save segments

DreamThing, 2020-12-21:
After a power cycle the Wemos D1 doenst remember the segments settings.

When the Wemos is powerd off and turnd back on, all the segments are gone and back to the default "Segment 0".
Could this setting save to the flash?

daromer2, 2020-12-21:
I think you need to explain a bit more what you done and not done or even use the template when reporting. 
Saving segments works fine here and i can save them. I have more than 10+ D1 running with many segments saved with latest fw and they are all saved.

Aircoookie, 2020-12-21:
Hi, of course you can save the segments :) Just create a preset with your preferred setup in the favorites tab, and set Boot preset to 1 in LED settings!

closed 2020-12-21 by Aircoookie