wled/WLED #1549: Filesystem not mounted

Michi1860, 2020-12-21:
After finally finding the lamp in wled I get more issues as you can see in the pictures below. The first message comes immediately after clicking on the lamp and the second one comes after klicking on anything in the app. 

I looked the issues up but there seems to be no answer. 

[image]
[image]

WoodyLetsCode, 2020-12-21:
What Microcontroller do you use? What binary file do you use?

Michi1860, 2020-12-22:
Controller: ESP8266 ESP-12F
Binary: WLED_0.11.1_ESP8266
Phone: XIAOMI NOTE 10 PRO

Aircoookie, 2020-12-22:
Hmm, it should work with your mcu, but just in case, try the `WLED_0.11.1_ESP8266_1M` binary

Michi1860, 2020-12-22:
I no longer recieve an error when using the ..._1M binary. Thank you very much sir.

Michi1860, 2020-12-22:
Though it doesn't seem to be working. I can press buttons in the app all I want but the lights are not changing. Yes, I solderd the right gpio2 which is D4 in this case. 
[image]

janbernhardt, 2020-12-22:
I had the same issue when uploading the source with Arduino. Finally I discovered that the FlashSize in my Arduino settings was set to `FS:none`. After changing it to FS:2MB it was working fine.
[image]

Michi1860, 2020-12-22:
This is my current FlashSize. I will try with different sizes now. 

[image]

WoodyLetsCode, 2020-12-22:
Use this: https://github.com/Aircoookie/WLED/wiki/Installation-using-the-official-GUI-tool-(ESP8266)

Michi1860, 2020-12-23:
Im beginning to think that the LED Strip is the issue. I orderd another one (same brand and type) and this time there is not a single LED emitting light when connecting the Strip to the designated 5V and GND. Even without any 'Data In' they should emit light, right?

Aircoookie, 2020-12-23:
@Michi1860 I would maybe check the data connection. Those jumper wires have almost no metal in them and I found them very hard to use reliably. Typically, addressable LEDs will NOT light up without a data input.

Michi1860, 2020-12-26:
In conclusion i ordered 5 different LED Strips. 2 of them apparently never worked. 1 Strip always showed the same (few) lights so it was broken too. The last 2 i ordered arrived at christmas and worked just perfectly fine. They even lighted up without any data connection. There is also to mention that i connected every single LED strip to the same setup. Therefore i can be sure that there is no error in the cata connection since the last 2 Strips worked just fine. 

What can we learn from this? 
Do not trust cheap sellers. 
Always double check your connections. 

Thanks for the support!

closed 2020-12-26 by Michi1860