wled/WLED #1554: Problem with resume WiFi connection

pepsonEL, 2020-12-21:
**Describe the bug**
Hi. I have on my ESP32 uploaded WLED 0.11.0 version. I have connected it to my WIFI with DHCP . But i have a problem.... Sometimes when WIFI is off for a some minutes but WLED(EP32) has still POWER, and then when WIFI come back, WLED no connected to WIFI. I must POWER OFF Wled and again POWER ON and it is connected. But it is no always problem. SOmetime when WIFI come back WLED connect automatically. But sometime no.... Please help....

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev] : ESP32
 - Version [e.g. 0.10.0, dev200603] : 0.11.0
 - Format [e.g. Binary, self-compiled] :BINARY

Hardwarekiller, 2020-12-21:
@pepsonEL 
You are using an old 11.0 firmware !
Install the current version 11.1 and test whether your problem has been resolved.

Otherwise I tested the old version 11.0 extensively.
The wifi operation never caused any problems, at least for me.

I think that this is more of a router problem or a hardware problem with the ESP32 because it has a very bad WiFi connection without an external antenna.
Therefore, an external antenna socket is often retrofitted there.
Or use an EPS32 version with an external antenna connection
An overloaded 2.4 GHz channel may also be the problem.
So for a test, change the WiFi channel
And also go to "Info" and look at the level values ​​under "Signal strength"

Hardwarekiller

pepsonEL, 2020-12-22:
Yes i update yesterday to version 0.11.1 and start test. I use ESP32 with external anntena. I have this:
https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcScmEf2xYquftIzxMYdlrS7EzMNn8ztPAxjGg&usqp=CAU

In "Info" show mi signal strange as 100%

Hardwarekiller, 2020-12-23:
If the display shows 100% it only means that the signal is maximally strong, but does not say anything about the quality of the signal.
As a test, change the WLAN channel, there may be too many neighboring WLANs on the same channel.
Or check the signal quality on the channel used with a mobile phone and an app, then change the channel and then test it.
Or set WLED to Access Point and log in with your mobile phone and see if the problem is still there

Hardwarekiller

pepsonEL, 2020-12-23:
No i not tested voltage during operation. 
Tell me also where i should plug capacitor as you write ? Parraller to power + and - ?

pepsonEL, 2020-12-23:
[image]

Hardwarekiller, 2020-12-24:
Yes, in parallel with the power supply of the EPS32.

Is the EPS32 supplied with power via the MicroUSB connection or via the connector strip of the ESP32 at pin VIN (5Volt)?

Do you operate the ESP32 and the LED strips on the same power supply, or are they separate?

Hardwarekiller

pepsonEL, 2020-12-24:
ESP32 is power from the same power supply as LED Strip. But ESP32 has power connected by VIN and GND pin. Power supply is 100W but my LED strip is about 4m and powers supply is more than consumption.

Hardwarekiller, 2020-12-28:
@pepsonEL 
Normally, the ESP is not supplied with the same power supply that the LED strip is attached to, and certainly not over 4 meters.

You should also consider the cable length
also take a few troubleshooting and stabilizing measures for the power supply to the ESP !!

Short-term current peaks in the LEDs cause brief voltage fluctuations at the end of the lead, and the thinner the lead, the greater the voltage loss on it.

Then the 5V voltage fluctuates at the Vin of the ESP32
This can / will lead to the EPS dropping out.
Also because the EPS has absolutely no smoothing capacitor.
The voltage at Vin should hardly fluctuate or with glitches
be contaminated.

Try not to operate the ESP32 via your power supply, but with an additional external power supply.
 
And whether the power supply has 100W or more says nothing about how impulse-proof it is or about the quality of it.
9 out of 10 cheap power supplies in Ebay / Amazon are no good for that.
Or you mistakenly buy a power supply unit that is not pulse-proof and is only intended for normal lighting, and then operates it without any sorting.
The difference is that the impulse fixed are far more expensive because they are technically more complex.
   
So a few measured values ​​of voltages and currents, cable lengths and diameters, and the number of LEDs, s

Hardwarekiller

pepsonEL, 2021-01-05:
Ok i done next test. I use for ESP external POWER by USB micro and other POWER Supply for led strip. And still is the problem. Yesterday i update my router firmware to newer and under update wifi was off and WLED on ESP32 dicsonect from wifi. AFter router update and reboot my WLED no connect again to the same wifi SSID. To today was disconect from wifi. When today POWER OFF and then POWER ON it was connect with no problem. 

In my opinion problem is when WLED lost connection wifi and can not resume this wifi.

ghpk, 2021-01-16:
Try to do Address Reservation in your Routers DHCP settings.
Also if your router has a settings for number of clients try raising it.
On some ISP provided router these issues come when number of connected devices exceed allowed clients.

closed 2021-06-09 by stale[bot]