wled/WLED #1556: Preset cycle time per effect reinit after reboot

cadavor, 2020-12-21:
After updating to 0.11.1, the parameters "Time per preset" of preset cycle configuration is lost after reboot.
It was previously saved to 90s before update since it roll back to 0.9s after every reboot.

It works well on 0.10 and 0.11.0

Reproduce :
Uncheck Preset cycle, set "Time per preset" to 90s, recheck preset cycle. It works well.
Reboot the ESP32 board (unplug, replug), the value is back to 0.9s (???)

Maybe it's due to the change : Fixed preset cycle default duration rounded down to nearest 10sec interval (#1458)

**WLED version**
 - Board: ESP32 dev
 - Version 0.11.1
 - Format : Binary

Thank you for your help!

Aircoookie, 2020-12-22:
Hi, by default the "Time per preset", as well as the other preset cycle parameters are non-persistant.
Try the following:
- Set the Time per preset to 90sec
- Check preset cycle and uncheck it again a few seconds later
- Go to LED settings, check "Save current preset cycle settings as boot default" and save.
Now the time should always be initialized as 90 sec correctly. Let me know if it works for you!

I am aware that this is quite involved and unintuitive, and I have plans to replace Preset cycle with a better system at some point :)

cadavor, 2020-12-22:
Thank you for the quick answer.

It works well now, but I would like to point out that the upgrade to version 0.11.1 change the value of time per preset by dividing it by 100.

closed 2021-06-16 by stale[bot]