wled/WLED #1557: LED freeze and bugs out

avanbaelinghem, 2020-12-22:
I just download the latest WLED (0.11.0) and it was working fine for like an hour or two, then then part of the lights froze and became unresponsive, while about the first 30 LEDS were responsive. So basically my lights are stuck with only the first 30 LEDS responding to commands on the app while the other 270 are frozen on a previous effect without even running the effect.

Aircoookie, 2020-12-22:
Hi! Can you check that the LED count in LED settings is still set to 300 and the brightness limiter milliamp setting is set to the rating of your power supply? Also make sure that the segment(s) cover the entire LED range.

Hope you can get your setup to work and merry Christmas :)

avanbaelinghem, 2020-12-24:
> Hi! Can you check that the LED count in LED settings is still set to 300 and the brightness limiter milliamp setting is set to the rating of your power supply? Also make sure that the segment(s) cover the entire LED range.
> 
> Hope you can get your setup to work and merry Christmas :)

It says a 17A power supply is connected but my power supply is only 8A. Would this cause the problem and if so how can i fix it?

sicknesz, 2020-12-24:
Hi aircoookie, first off excellent job on wled ! it's impressive, but im having almost the same issue, flashed the latest 0.11.1 on an esp8266 yesterday, worked fine for a few hours, retried today and it was a mess, effect in wled are fine but as soon as i start hyperion the led start acting weird and the esp reboot / crash (rst cause: 2 / rst cause: 02f00) my power supply is 8.5A, should i set the brightness limiter to 8A, i've already set the LED type to 12v (ws2811 12v 60led/meter 5 meters long 90w total) im using a 100w-12v-8.5A as power supply, i've read what you suggested to avanbaelinghem and having the proper setting would be a good start. from itself your app as set 850mA if a recall correctly (8.5A, 850mA) should i add a 0 ? 

Also, to be sure it wasn't my hyperion missing things up i've reflashed a different client (hyperion_rgb_controller, which is using fastled in the background) and i have no issue at all work perfectly fine, only issue is when i use wled as client so my guess is that it's either a misconfiguration or there's something going on with the received data.

thedeany, 2020-12-28:
I think I'm encountering this issue as well, if not something similar. I've only noticed it when I set it to my "full white" mode. After setting it to that, after a while the LEDs all turn off (a few at the start of the strip might still be on), the LED onboard the ESP8266 is off, and WLED seems to be unresponsive for a time. A few minutes later I can usually resume using WLED.

I did set my Maximum Current setting lower because I was getting some discoloring when it was set to my actual PSU rating, so I set the max current to the highest setting where I no longer got discoloring. Is this causing the issue?

Let me know if there's any further troubleshooting details you need!

anastasiuspernat, 2021-02-12:
I'm encountering the same problem. After a while about 50 LEDs on the end of any strip that I have become unresponsive. I have strips with 242 LEDs, the strip length set in the Config of WLED. 

It starts to glitch after some some time, like 1-2 hours. It happens more frequently when using LedFx, or increasing brightness or switching off brightness limiter. And LedFx causes the whole strip to freeze even on the default brightness settings.

Also when it happens WiFi may become unresponsive so I have to unplug and re-plug my ESP8266. Though some times it doesn't. Can it be the hardware problem, like overheating (the board doesn't feel too hot though)? I also tried different power sources - didn't help.

Can provide any troubleshooting details as well! Thank you.

sansillusion, 2021-02-12:
This sounds like power issues. Make sure you have good stable power supply
strong enough for that amount of LEDs. Also power injection is important on
long strips.

closed 2021-06-28 by stale[bot]

Error-Gap, 2024-02-12:
Having the same issue, but only after updating from WLED 0.14.0b1 to WLED 0.14
On the previous version, the WLED device works happily with LedFX. After the update, the dynamic lighting updates seem to overwhelm the device, ping gets crazy long and it's unresponsive until LedFX is disabled

I don't think it's an LedFX or power issue as it worked fine on the previous version.

blazoncek, 2024-02-12:
That has been fixed in 0.15 I think.