wled/WLED #1558: LEDs do not light up

3DFans, 2020-12-22:
Hey
I've connected and connected everything so far. The web interface is also available, but I can do what I want, but no LEDs light up. I put the datapin on D4, but without success. The LEDs stay off. Does anyone else have a tip?

Aircoookie, 2020-12-22:
Hi! How many LEDs do you have and how are they wired up? What LED count and brightness limiter setting do you have in LED settings? You can send pictures of your setup, that often helps with figuring out what the issue is :)

3DFans, 2020-12-22:
Hey, I've attached a few photos. The whole thing is supplied by a power bank with 5V, 2.4A.
[image]
[image]
[image]

sansillusion, 2020-12-22:
Mabe you have the wrong pin ? I did a lamp yesterday and I soldered the
data wire to D4 on my board, the lights would not turn on, so I decided to
try pin D2 (chinese esp32 devboard) and tada ! Lights turned on ! Mabe just
a different labeling sheme.

3DFans, 2020-12-22:
> Mabe you have the wrong pin ? I did a lamp yesterday and I soldered the data wire to D4 on my board, the lights would not turn on, so I decided to try pin D2 (chinese esp32 devboard) and tada ! Lights turned on ! Mabe just a different labeling sheme.

That's what I thought and tested the pins, but none of them bring success. : /

huggy-d1, 2020-12-23:
Swap out the Dupont jumper(s) after finding which one is not crimped properly.

What is the voltage from GND to +5V on the LED strip?

Wiggle the dupont jumpers while measuring.  Any change?

closed 2021-06-16 by stale[bot]