wled/WLED #1560: LEDs flicker in random intervals

LeoLTM, 2020-12-22:
I have about 2,7m of WS2812b LEDs (60LEDs/m) on an ESP8266 and the second half of them flickers for a half of a second randomly and I don't know what causes this problem.
The setup is as following:
The 2,7m are cut in half to mount them in a 90 dregree angle. The first half of the LEDs gets power and data from the ESP and the data wire goes out at the end to the next half of the LEDs, but the power doesn't. There is a second power input after the cut. 
Everything is powered by a Meanwell 10A powersupply. I made a short video but I don't know where to upload it (is there something like pastebin for videos?)

I hope there is a solution for this

Legsmaniac, 2020-12-22:
> I made a short video but I don't know where to upload it

Right here in your reply window.  Just drag and drop video into it.  
See the "New" note bottom of your reply window.  😊

LeoLTM, 2020-12-22:
I wasn't able to use this feature because the file is too big (~35MB) and GitHub only allows 10MB, but thanks for the answer!

huggy-d1, 2020-12-23:
If I were to guess, based on this phrase "first half ... gets power ... from ESP", the ESP ground and the LED ground(s) are not interconnected in some way.
It could also be the 90 degree bend interconnection.  I believe if this were the problem, you would have already found/fixed it.
It could be there is a rogue segment (#1) in there causing problems.  Configure segment 1 so the start / stop are 0, then save the segment.  It should delete that segment.  If that was the problem, the strip 2nd half should change in some way when the segment gets deleted.

Try adding a 2nd segment (#1), for say 10 LEDs, and see if that segment works.

How many total LEDs are configured in WLED? The default is 30, and for sure the first half of the strip has more than 30, right?

closed 2021-06-16 by stale[bot]