wled/WLED #1561: Would like to be able to change the start point of a segment

coryking, 2020-12-22:
My particular project requires me to change the first LED of each segment to be something other than the first element in the segment.

Why? Well, I'm putting these lights into a bunch of hexagons. Each hexagon is a segment. Unfortunately due to my oversight, the "zero" element of each segment is not the same physical location on each hexagon. I would like it for the software to let me set the "first" element to be the bottom of each hexagon regardless of where the actual location is. This way the effects can be uniform across all hexagons.

I could see this feature being used by anybody who is driving a bunch of segments where the physical start of a segment isn't necessarily where the effects should originate from.

In this shot of my project, I've circled the start of each segment in yellow and where I would like the software to think the "first" element should be.
[image]

I plan to write this feature and before I do, I'd love any input as to how I might do it in a way that it could be merged back into master.

huggy-d1, 2020-12-23:
A recent request asked for a virtual to physical LED mapping option.

@Aircoookie said it's available in the WLED code already, in /FX_fcn.cpp, right near the top.  You can set the actual sequence, compile, download and go.

coryking, 2020-12-23:
Awesome! I'll give it a shot and see if it meets my needs!

Aircoookie, 2020-12-24:
Hi!
That feature would be absolutely great to have - you are also not the first one who has suggested it (https://github.com/Aircoookie/WLED/issues/788)

Adding this capability in a way it could be merged would require:
- Adding an extra `uint16_t` with the start index to the segment object
- Writing extra code in setPixelColor() and getPixelColor() to take the offset into account (TBD: how should it affect Grouping and Spacing? I believe it should have higher priority, so if the start LED is 2 and grouping 5, LEDs 2-7 should be the first "virtual" LED)
- Adding the extra property to the JSON (and maybe later HTTP) API
- Modifying the UI to add a 5th imput field to the segment control

Let me know if you'd like to contribute to (some of) these and let's make it happen 😄

Aircoookie, 2021-06-30:
Finally done :)

closed 2021-06-30 by Aircoookie

Gabs64, 2023-08-10:
hello, just wondering. where do you find FX_fcn.cpp?

softhack007, 2023-08-10:
> where do you find FX_fcn.cpp?

in `wled00/` folder. 
https://github.com/Aircoookie/WLED/blob/main/wled00/FX_fcn.cpp

The mentioned solution is called "Segment Offset" - use it to rotate effects on a ring (or hexagon or other round stuff)
[image]