wled/WLED #1568: RGB GRB leds together

t0mb3r, 2020-12-23:
Hi, is there a way to set RGB and GRB leds together? 
I bought some smd leds and they sent me mixed up and I cant get to sync them.

Aircoookie, 2020-12-24:
Hi!
Not really - but there are two ways you can mitigate it:
- The easy way: Make two segments and set the LEDs with the "wrong" order to the color that matches (e.g if you want red you set GRB to red and RGB to green). This will only work with Solid or really simple effect modes though.
- The better way: I've actually run into exactly the same issue two weeks ago, so I made a code fix :) It needs to be custom compiled in however, there is no runtime setting for it. This is the line where the defines start to set the color order you want and the range https://github.com/Aircoookie/WLED/blob/master/wled00/NpbWrapper.h#L44 :)

t0mb3r, 2020-12-24:
So if I have led 0-10 as RGB and from that on its GRB. I set:

//enable color order override for a specific range of the strip
//This can be useful if you want to chain multiple strings with incompatible color order
//#define COLOR_ORDER_OVERRIDE
#define COO_MIN    0
#define COO_MAX   10 //not inclusive, this would set the override for LEDs 0-26
#define COO_ORDER COL_ORDER_GRB

How to I compile it? 

Thank you

t0mb3r, 2020-12-25:
Tried following this guide, but I couldn't do it. 

https://github.com/Aircoookie/WLED/wiki/Compiling-WLED

t0mb3r, 2020-12-25:
Done! It works thank you!!

closed 2020-12-25 by t0mb3r