wled/WLED #1570: Sliders 0 to 255 setting

Legsmaniac, 2020-12-24:
Apologies if this has already been mentioned/suggested and/or if it's something that is planned for the future.

I notice how a small box pops up to the right of the Speed and Intensity slider controls showing their position from 0 to 255 but not so for the RGB sliders nor the Brightness.    I think it would be nice if all the sliders had the same feature as the Speed and Intensity sliders and hope you can/will consider this for future inclusion.

Muchas Gracias.

tonyn0, 2021-01-20:
This would be great.

andyshinn, 2021-01-20:
I'll take a look at this soon as I initially added it. It was only for those initially to see how the feature would look. But it can be added to the others pretty easily.

blazoncek, 2022-10-10:
Available in 0.14

closed 2022-10-10 by blazoncek