wled/WLED #1572: Power measurement sensors?

fribse, 2020-12-24:
**Is your feature request related to a problem? Please describe.**
I measure power on a lot of devices around the house, and it would be nice to have the WLED lights report in power consumption as well. I know it's not going to be accurate, but it would be nice to have a best guestimate included.

**Describe the solution you'd like**
A way to set a 'base' powerusage for when it's turned off (this would have to be measured externally by the user), and then have wled calculate current usage from that and current light usage.
Sensor reporting current kW.
Sensor reporting kWh accumulated over the day (from midnight to midnight).

**Describe alternatives you've considered**
I could add a measuring plug to all the installations, but that seems a bit excessive.

daromer2, 2020-12-24:
The meassured (estimated) value already exist in wled
[image]

Could send that data over MQTT as an example. If you do like i do where all such data is stored in InfluxDB you could easily get Wh calculated :)  

Sending the info data wouldnt cost that much in the code itself. No extra calculations needed.

fribse, 2020-12-24:
Yes, it's absolutely possible to have this info calculated in Home Assistant with an integration sensor, but it would still be nice to have it straight from the horses mouth.

gothdawid, 2025-01-24:
We know that based on the 'estimated current' and a fixed voltage for the LEDs, it's possible to calculate power consumption, but it would be nice to integrate such devices without additional modifications (e.g., in Home Assistant).

Additionally, the 'Estimated current' returns an incorrect value (a constant 450mA) when the LEDs are turned off (using a relay).

It would be a great option if WLED could return the actual energy consumption based on the selected LED Voltage * Estimated current.

gothdawid, 2025-01-24:
Additionally, a better option for calibrating the energy usage of individual colors would be useful. For example, the red color consumes 16W, green 15.5W, and blue 16.5W, but when white is used, it consumes only 13W. In such a case, the limiter doesn't work correctly.

closed 2026-04-17 by DedeHai