wled/WLED #1573: BitFocus Companion integration

christiansass, 2020-12-25:
I would love for there to be a simple way to integrate or trigger presets from Bitfocus Companion via my Elgato Streamdeck.. Basically, there's an onAir sign developed by "make that inc" out of Portland, that I would like to trigger when going live
The developer insists there's a way to make this work... it's just not obvious or simple.. I also have a request to integrate this over at the companion forums  I thought id open a request here to see if I could get pointers from any of you to help get this working

https://makethatinc.com/collections/signs
https://bitfocus.io/

Aircoookie, 2020-12-25:
Hi!
I don't know about BitFocus, but I do have a stream deck as well and use the `System -> Website` icons with `Access in background` checkbox enabled to send HTTP API commands to load my presets :) I would assume Companion to support similar functionality to interface with either my HTTP or JSON API (both documented in the wiki)

Merry Christmas!

christiansass, 2020-12-28:
read the wiki

https://github.com/Aircoookie/WLED/wiki

set a static IP can you give me an example of a command you use?

here's where I am so far...
[image]

flow1986, 2023-05-09:
https://github.com/bitfocus/companion-module-requests/issues/335

would be so nice to control (with feedback) Wled from Companion