wled/WLED #1574: I thought it was WS 2812

bagotu, 2020-12-25:
I bought this garland. It works with the Chinese controller. With WLED in any way ... Maybe you have any ideas how to connect WLED to it?

https://aliexpress.ru/item/1005001476385975.html?spm=a2g0s.9042311.0.0.794833edT4HYlS

Aircoookie, 2020-12-25:
Sorry, unfortunately the listing gives no clue as to what driver chip or protocol these use.
We would need some one with these, an oscilloscope and a lot of time to reverse engineer their protocol

lucaf86, 2020-12-26:
I'm using similar led bought on amazon https://www.amazon.it/gp/product/B08GG5QFRH/ref=ox_sc_saved_title_6?smid=ABFJ1VGQJGWEF&psc=1 on my tree.
They work with Wled like normal ws2812 but they are a bit strange.
I need to add a level shifter from 3.3v to 5V. Then if you put the resistor on data line try to remove it.My led seems to wor only with no resistor or with a very low value like no more than 100 or 150 ohm.
I have 3 string and I tried to connect them in series by injecting power on each of them to avoid power loss but the second and third string didn't work as expected. It looks like if they where in parallel, so the first led of each string behave the same but they were in series.... not really understand why. For sure they are not regular ws2812.
In the eand I used an esp32 and connevted each string on a different pin using the multistrip usermod.In this way I can use them as a regular 300 led series string.

closed 2021-06-16 by stale[bot]