wled/WLED #1582: bugs with commands to mqtt/api

daddvok, 2020-12-27:
I catch very strange bugs when sending to a topic /api, for example FX=0 switches to "Solid", and &FX = 0 switches to Effect number 6. And vice versa. that is, somewhere & FX teams work, and somewhere FX. And the same watch in other api commands, whether &A, &IX, &SX. 
**Example 1**
[image]
**Example 2**
[image]

Aircoookie, 2020-12-27:
Hi! Stock WLED does not return such JSON to the `.../v` topic. Furthermore there isn't even an effect 144. Are you using a fork or usermod?

daddvok, 2020-12-27:
> Hi! Stock WLED does not return such JSON to the `.../v` topic. Furthermore there isn't even an effect 144. Are you using a fork or usermod?

Hi, I just reformatted the title of the topic for better understanding. The value is not formatted in any way

daddvok, 2020-12-29:
@Aircoookie I connected and flashed another esp, I send both the same commands, everything works fine on the new one, on the old one - no, what could be the problem?

daddvok, 2021-01-04:
The reason was a bad power supply for the LED strip, it apparently gave false pulses, after replacement everything works fine

closed 2021-01-04 by daddvok