wled/WLED #1584: Variables not declared in function "void addToJsonState(JsonObject& root)"

gsieben, 2020-12-28:
I am currently building a usermod with the full integration of relays. For this I use among other things functions like:

void addToJsonInfo(JsonObject& root)
void addToConfig(JsonObject& root)
etc. 

Now I came across a little problem which I haven't understood yet. If I use variables of my usermod inside the functions, it works quite well. No matter if they are private or global. But for a reason I don't know yet, it doesn't work with: void addToJsonState(JsonObject& root). There the variables are shown as not declared. 

Can anyone help me with this?
Ceers
Gabriel

Aircoookie, 2020-12-28:
Hi!
That is strange, it should work fine.
Some things to check:
- Is the `addToJsonState()` method inside your usermod class?
- Is the variable shown as not declared in your usermod class?
- Is the declaration of the variable above the method in the file? This shouldn't matter in classes but you never know.

If it doesn't solve the problem, feel free to share what you have so far and I'll take a look if I can spot what the culprit is :)

gsieben, 2020-12-28:
I used the usermod_v2_example.h. Yes to all your questions.

[image]

gsieben, 2020-12-28:
#pragma once

#include "wled.h"

/*
 * Usermods allow you to add own functionality to WLED more easily
 * See: https://github.com/Aircoookie/WLED/wiki/Add-own-functionality
 * 
 * This is an example for a v2 usermod.
 * v2 usermods are class inheritance based and can (but don't have to) implement more functions, each of them is shown in this example.
 * Multiple v2 usermods can be added to one compilation easily.
 * 
 * Creating a usermod:
 * This file serves as an example. If you want to create a usermod, it is recommended to use usermod_v2_empty.h from the usermods folder as a template.
 * Please remember to rename the class and file to a descriptive name.
 * You may also use multiple .h and .cpp files.
 * 
 * Using a usermod:
 * 1. Copy the usermod into the sketch folder (same folder as wled00.ino)
 * 2. Register the usermod by adding #include "usermod_filename.h" in the top and registerUsermod(new MyUsermodClass()) in the bottom of usermods_list.cpp
 */

//class name. Use something descriptive and leave the ": public Usermod" part :)
class MyExampleUsermod : public Usermod {
  private:
    //Private class members. You can declare variables and functions only accessible to your usermod here
    unsigned long lastTime = 0;
  public:
    //Functions called by WLED

    /*
     * setup() is called once at boot. WiFi is not yet connected at this point.
     * You can use it to initialize variables, sensors or similar.
     */
    void setup() {
      //Serial.println("Hello from my usermod!");
    }

    /*
     * connected() is called every time the WiFi is (re)connected
     * Use it to initialize network interfaces
     */
    void connected() {
      //Serial.println("Connected to WiFi!");
    }

    /*
     * loop() is called continuously. Here you can check for events, read sensors, etc.
     * 
     * Tips:
     * 1. You can use "if (WLED_CONNECTED)" to check for a successful network connection.
     *    Additionally, "if (WLED_MQTT_CONNECTED)" is available to check for a connection to an MQTT broker.
     * 
     * 2. Try to avoid using the delay() function. NEVER use delays longer than 10 milliseconds.
     *    Instead, use a timer check as shown here.
     */
    void loop() {
      if (millis() - lastTime > 1000) {
        //Serial.println("I'm alive!");
        lastTime = millis();
      }
    }

    /*
     * addToJsonInfo() can be used to add custom entries to the /json/info part of the JSON API.
     * Creating an "u" object allows you to add custom key/value pairs to the Info section of the WLED web UI.
     * Below it is shown how this could be used for e.g. a light sensor
     */
    /*
    void addToJsonInfo(JsonObject& root)
[... 56 lines cut ...]
     * This is called by WLED when settings are loaded (currently this only happens once immediately after boot)
     * 
     * readFromConfig() is called BEFORE setup(). This means you can use your persistent values in setup() (e.g. pin assignments, buffer sizes),
     * but also that if you want to write persistent values to a dynamic buffer, you'd need to allocate it here instead of in setup.
     * If you don't know what that is, don't fret. It most likely doesn't affect your use case :)
     */
    void readFromConfig(JsonObject& root)
    {
      JsonObject top = root["top"];
      userVar0 = top["great"] | 42; //The value right of the pipe "|" is the default value in case your setting was not present in cfg.json (e.g. first boot)
    }

   
    /*
     * getId() allows you to optionally give your V2 usermod an unique ID (please define it in const.h!).
     * This could be used in the future for the system to determine whether your usermod is installed.
     */
    uint16_t getId()
    {
      return USERMOD_ID_EXAMPLE;
    }

   //More methods can be added in the future, this example will then be extended.
   //Your usermod will remain compatible as it does not need to implement all methods from the Usermod base class!
};

gsieben, 2020-12-28:
#pragma once

#include "wled.h"

#define MAXRELAYS 4 // Dependend from Alexa Max Devices

/*
 * Usermods allow to controll relais over http/weppage/app/alexa
 */
void InitRelais(); 

class UsermodGeoGab : public Usermod {
  private:
  
  uint8_t relaysno;

  struct relay_t {
    String name;
    uint8_t gpio;
    bool sactive;
    bool invert;
    bool status;
  } relays [MAXRELAYS];

  public:
    void setup() {
      Serial.print(F("GeoGab-Relays: Starting UserMod ID: "));
      Serial.println(getId());
      InitRelais();                         /* Initialization of the settings */
      // TODO: Alexa
      // TODO: MQTT

      /* TODO: Server.on for /Relay? calls. Just ON/OFF but not setup*/

    }

    void loop() {
      /* No necessary yet */
    }

    void readFromJsonState(JsonObject& root)
    {
      /* Structure of json setup (example): {"relays":{"no":3,"name":["Dev1","Dev2","Dev3"],"gpio":[16,5,13,15],"sactive":[0,0,0,0],"invert":[0,0,0,0]}}*/
      
      /* Structure of Json change status (example): {"relays":{"switch":[0,0,0,0]}}, {"relays":{"toggle":[0,0,0,0]}}, {"relays":{"on":[0,0,0,0]}}, {"relays":{"off":[0,0,0,0]}} 
          * switch (1=on, 0=off)
          * toggle (0=unchanged, 1=toggle) 
          * on (1=on, 0=unchanged)
          * off (1=off, 0=unchanged)
      */

      if (root[F("relays")]) {
        if ( root[F("relays")][F("no")]) {
          /****** Set new settings ******/
          Serial.println(F("GeoGab-Relays: readFromJsonState - Function: Setup"));
            /* Get Values */
            relaysno = root[F("relays")][F("no")];
            for (uint8_t i = 0 ; i<relaysno; i++) {
              relays[i].name=root[F("relays")][F("name")][i].as<String>();      // Device Names
              relays[i].gpio=root[F("relays")][F("gpio")][i];                   // Relays PIN
              relays[i].sactive=root[F("relays")][F("sactive")][i];             // Boot Status
              relays[i].invert=root[F("relays")][F("invert")][i];               // Invert
            }
            InitRelais();           /* Initialization of the settings */
            serializeConfig();      /* Save to Config */

        } else {
          if (relaysno==0) return;      // Means no relays defined yet
          /***** Functions ******/         
          if (root[F("relays")][F("switch")]) {
[... 120 lines cut ...]
          rroot[relays[i].name] = relays[i].status;
        }
      } else {
       root[F("relays")] = "No relay has been configured yet.";
      }
    }
};

    /* Automatic: Adds the relay states to json/state */
    void addToJsonState(JsonObject& root)
    {
      JsonObject top = root.createNestedObject("relays");
      top[F("no")] = relaysno;
      JsonArray name = top.createNestedArray("name");
      JsonArray gpio = top.createNestedArray("gpio");
      JsonArray sactive = top.createNestedArray("sactive");
      JsonArray invert = top.createNestedArray("invert");
      for (uint8_t i = 0 ; i<relaysno; i++) {
        name.add(relays[i].name);
        gpio.add(relays[i].gpio);
        sactive.add(relays[i].sactive);
        invert.add(relays[i].invert);
      }

    }

gsieben, 2020-12-28:
Gosh. I missed the bracket at the end and was under my procedure. I was just blind. Thanks a lot

closed 2020-12-28 by gsieben