wled/WLED #1587: Deep sleep and power bank

t0mb3r, 2020-12-28:
Hi, I've been searching for an answer and was unable to resolve it. 

Problem: WLED goes offline after some time (more than an hour I believe).

I'm using a powerbank and a nodemcu. Running latest update. WIFI sleep is disabled. 
Tried with 3 different power banks. No luck. 

Where can I disable deep sleep? Or is there anything else I can try?

Thank you

Aircoookie, 2020-12-28:
Hi!
Most powerbanks have this feature where they turn off if the connected device draws too little current.
WLED never enters deep sleep, it isn't able to do so and still be permanently available for commands.
Still, the controller only draws <100mA which might not be enough to keep the power bank awake.
The only way to fix that problem is to artificially increase the power consumption is to leave the LEDs on (in a dimmed down state). Of course that way your powerbank won't last for long, but it has been designed for charging devices, not for powering low-current stuff for extended periods.

Typically this low-current shutoff happens after several seconds to minutes though, not after some hours. Either your power bank has been shutting off because it's empty or the NodeMCU continues to receive power but drops off your WiFi for a different reason. Can you try running the controller on a wall charger or PC USB port for several hours to see if that also causes WLED to go offline?

Hope we can find a solution :)

gsieben, 2020-12-28:
Unfortunately, that's the way it is with powerbanks. I've been looking for a powerbank that doesn't switch off for a very long time. A year ago I installed my son a LED matrix that goes into deep sleep and wakes up through vibration sensors. And with that, unfortunately, the powerbank turns off at some point. Unfortunately, to this day I have no solution except to "waste" power again and again on purpose.

Tech-TX, 2021-05-27:
I have a ROMOSS powerbank that stays alive with a 35mA load... haven't tried it in Deep Sleep though.  Once it drops down to 20uA I'd almost expect it to turn off, as it has a 1 minute timer for auto shut off.  I also have a cheap no-name power bank with NO built-in smarts that won't shut off as long as the switch on the power bank is on and there's sufficient power in the cells.

closed 2022-04-16 by stale[bot]

Dim3390, 2023-06-16:
Another solution is to use another ESP with sleep mode and PIR or other other sensors  that uses a relay to switch
 power to ESP with WLED installed.