wled/WLED #1590: No WLED-AP on Sonoff Basic R1

FulvioSpelta, 2020-12-29:
**Describe the bug**
After a fresh flash no WLED-AP is available

**To Reproduce**
Flash WLED_0.11.1_ESP8266_1M.bin on a sonoff basic R1

**Expected behavior**
Normal availability of the WLED-AP to start configuration and use

**WLED version**
 - Board: sonoff basic R1
 - Version 0.11.1 (checked with WLED_0.10.2_ESP01_1M_full_fix2009030.bin too)
 - Format Binary

**Additional context**
Following the hint in the #461 I've compiled wled myself using [env:esp01_1m_full]
Source from v 0.11.1 zip (https://github.com/Aircoookie/WLED/archive/v0.11.1.zip)

Modifications in platformio.ini
Commented
`; default_envs = travis_esp8266, travis_esp32`

un commented
`default_envs = esp01_1m_full`

First test: compiled without any change to have a bin that should be as the downloadable version
Result: same problem no AP

Second test: compiled adding all debug flags:
```
[env:esp01_1m_full]
board = esp01_1m
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} ${common.debug_flags} -D WLED_DISABLE_OTA
```
Result: no AP or intermittent SSID emission but without any success trying to join it using my phone.

Following the debug log captured:
```
r$l▒▒▒▒$▒|
▒
$▒b|▒▒▒▒r▒blb▒▒on▒lon▒▒#bp▒▒l{l{lp▒n▒
▒
$
▒▒
#
o▒|▒▒
▒c▒▒o▒l▒▒$`▒o
$`os▒▒▒N
#l r▒▒▒
b$`▒{$▒▒|▒l$▒l▒N▒
---WLED 0.11.1 2012180 INIT---
esp8266 2_7_4_7
heap 39848
Preset file not found, attempting to load from EEPROM
deEEP complete!
Reading settings from /wsec.json...
Restore settings from EEPROM
EEPROM settings invalid, using defaults...
Writing settings to /wsec.json...
Writing settings to /cfg.json...
Reading settings from /cfg.json...
Ada
No connection configured. Opening access point WLED-AP
Init AP interfaces
---DEBUG INFO---
Runtime: 10000
Unix time: 10
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 6428
---DEBUG INFO---
Runtime: 20003
Unix time: 20
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7703
---DEBUG INFO---
Runtime: 30006
Unix time: 30
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7637
---DEBUG INFO---
Runtime: 40009
Unix time: 40
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7619
---DEBUG INFO---
Runtime: 50012
Unix time: 50
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 60015
Unix time: 60
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 70018
Unix time: 70
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 80021
Unix time: 80
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 90024
Unix time: 90
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 100027
Unix time: 100
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 110031
Unix time: 110
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 120035
Unix time: 120
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 130039
Unix time: 130
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 140043
Unix time: 140
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
---DEBUG INFO---
Runtime: 150047
Unix time: 150
Free heap: 33840
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7620
Connected AP clients: 1
---DEBUG INFO---
Runtime: 160051
Unix time: 160
Free heap: 32968
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7617
---DEBUG INFO---
Runtime: 170055
Unix time: 170
Free heap: 32776
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7680
---DEBUG INFO---
Runtime: 180059
Unix time: 180
Free heap: 32776
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7692
---DEBUG INFO---
Runtime: 190063
Unix time: 190
Free heap: 32776
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7702
---DEBUG INFO---
Runtime: 200067
Unix time: 200
Free heap: 32776
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7703
---DEBUG INFO---
Runtime: 210071
Unix time: 210
Free heap: 32776
Wifi state: 0
State time: 10000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 7704

```

I'm available for other tests
Thanks

DeviousPenguin, 2020-12-30:
I would be tempted to try flashing 0.10.x to devices with 1MB of flash.

v0.11.x use the new 'littlefs' method to store presets, and may not work too well with less flash memory.

However if you do manage to get it working with 0.11.x then please share how you pulled it off. :)

FulvioSpelta, 2020-12-30:
> I would be tempted to try flashing 0.10.x for devices with 1MB of flash.
> 
As reported I've checked the  WLED_0.10.2_ESP01_1M_full_fix2009030.bin too, but same ptroblem.

In the meanwhile (due to some posts found googling) I've checked the WLED_0.9.1_ESP8266_1M_full.bin with no success.

I'm doing other tests, I'll report the results as soon as possibile.

FulvioSpelta, 2020-12-30:
Hi all, an update: reading soma posts i've noticed that almost all the successfull installation
refer to OTA version. I've compared the 2 versions finding:

.bin compiled myself using source v0.11.1 untouched exept the following tasks

**Task version 1 (1Mfull plus debug without OTA)**
```
[env:esp01_1m_full]
board = sonoff_basic
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} ${common.debug_flags} -D WLED_DISABLE_OTA
```
Result: no AP at all

**Task version 2 (1Mfull plus debug with OTA)**
```
[env:esp01_1m_full]
board = sonoff_basic
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} ${common.debug_flags}
```
Result: AP visibile, connected, capitve portal ok, wifi configured (I'm finding problems about connection in STA mode but this will be investigaed more).

This test seems show that there's some differences between OTA and noOTA activating the AP.

I've compared the 2 logs but, ignoring the messages generated by the use of captive portal, it seems that there's no differences:
[image]
[image]
[image]

Attached both logs.

[sonoff_basic_test.zip](https://github.com/Aircoookie/WLED/files/5755615/sonoff_basic_test.zip)

FulvioSpelta, 2020-12-31:
Using WLED_0.8.6_ESP01_1m_ota it's all ok (AP visibile, connected to it, wifi setup, wifi sta mode connection, UI available and responsive.

Not tested driving led because i need to build a level shifter.

I'd really like to help to let newer versions successfully running but currently i need hints about how to go on.

DeviousPenguin, 2020-12-31:
It may be worth doing an 'erase_flash' using ESPTool before flashing the older firmwares, it may help the settings get stored.

I'm not really sure what else to suggest as I've not managed either to make WLED work on the Sonoff Basic R1, I only tried ot briefly as an experiement before flashing it back to Tasmota. I had the exact same experience with v0.11.0 wher e the 'WLED-AP' SSID would briefly appear before vanishing.

This youtube vid shows a Sonoff basic apparently running WLED but it is R2 variant.

FulvioSpelta, 2021-01-01:
Hi all, I've noticed that from 0.8.6 and 0.11.1 the esp8266 cores have been changed, so I've done the following tests:

**Environment**
Source: 0.11.1 untouched
Added to platformio.ini the old cores:
```
arduino_core_2_4_2 = espressif8266@1.8.0
arduino_core_2_5_2 = espressif8266@2.2.3
```
**Compiled and tested the following env (synthesis):**
[image]

_Currently it seems that changing the cores let the 0.11.1, with some functions disabled to save space, to run properly.
For my own use case the T8 setup is good enought so I'll stop here my test and will go further with level shifter and real led driving._

I'm available for support and/or tests if someone would like to go deeper on this topic.
I  won't publish any source because the only changes are documented in the details section.
Attached you can find all the compiled .bin as reference.
[0.11.1_cores.zip](https://github.com/Aircoookie/WLED/files/5759119/0.11.1_cores.zip)

ENV Details
T1
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_OTA
```

T2
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m0m}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_INFRARED -D WLED_DISABLE_MQTT -D WLED_DISABLE_WEBSOCKETS
```
**Due in T2 test the errors in UI seems about the lack of gilesystem to store the presets, starting from the next T3 I've always used the ldscript_1m128k**

T3
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_INFRARED -D WLED_DISABLE_MQTT -D WLED_DISABLE_WEBSOCKETS
```

T4
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_WEBSOCKETS
```

T5
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_INFRARED -D WLED_DISABLE_WEBSOCKETS
```

T6
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_MQTT -D WLED_DISABLE_WEBSOCKETS
```

T7
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_INFRARED -D WLED_DISABLE_MQTT -D WLED_DISABLE_WEBSOCKETS  -D WLED_DISABLE_OTA
```

T8
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_WEBSOCKETS  -D WLED_DISABLE_OTA
```

closed 2021-01-01 by FulvioSpelta

joshlovell15, 2021-04-23:
@FulvioSpelta I was facing the same issues as you, anything above V8.5 of WLED would not run for me. I managed to use T8.bin to get my first one up and running. I was wondering if you could assist with getting V12.0 using the correct cores? Love your in-depth research!!

FulvioSpelta, 2021-04-25:
Hi @joshlovell15 sorry for the late answer. Currently i've modified and tested only the Mirai 0.11.1 version.
Had you try to change the core and compile the 0.12?

jfeuceda, 2021-06-09:
Hi Fulvio,
Thanks for that very well documented case, sure mine is a very dumb question, but your compilation work with GPIO14? or did you do any hardware mod to get GPIO2?
Thanks in advance.

FulvioSpelta, 2021-06-11:
Hi jfeuceda,
you are right, good question to completely fix the topic. :-)

My experience is:
GPIO14 never run correctly: the fw return unstable and unusable.
To avoid hw modding i tested the use of GPI03 (RX) that is directly available.

This is the env definition i've used:
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_WEBSOCKETS  -D WLED_DISABLE_OTA -D LEDPIN=3
```

I suggest to use an IC level shifter, as far as y know (and in my tests) the 2N7000 that I've seen often suggested is a bit too slow.

What currently is missing in my experience is a well done stress test (continous use for long time).
Pls share your experience :-)

jfeuceda, 2021-06-15:
Hi Fluvio,

I just flashed my old sonoff basic R1, and WLED-AP, working perfectly, thanks bro it really helps me.

Regards.

FulvioSpelta, 2021-06-18:
Happy to have been helpful :-)