wled/WLED #1592: [WLED-IP]/edit show nothing

Bebick, 2020-12-29:
Hey guys,
i tryed to save and restore my presets based on the instructions of [https://github.com/Aircoookie/WLED/wiki/Presets](url) but the site of [WLED-IP]/edit (in my example 4.3.2.1/edit) doesn's show me anything.
What am I doing wrong?

I use a WEMOS D1 mini pro with WLED Software 0.11.1.
The "Lock wireless (OTA) software update" Button is deactivated.

Bebick, 2020-12-29:
[image]

Aircoookie, 2020-12-29:
Hi!
I never ran into that problem before... Do you have a working filesystem (no Error 10 in the main UI)? You can also try a different browser or accessing the /edit page not via the accesspoint but when connected to WiFi.

twlare, 2020-12-30:
@Bebick /edit opens a file editor, but does not select a file. Do you see the files listed on the left side? You should see "/cfg.json" and "/presets.json". If you click on "/presets.json" do you see the file in the edit window? A new installation with no presets configured will show "{"0":{}}" on line 1.

Bebick, 2021-01-07:
Sorry for the late response.
The filesystem works, i don't have any Errors in the main UI.
I tryed Mozilla Firefox, Internet Explorer and Microsoft Edge - always the same problem.

@twlare i don't see anything. only the dark gray line. 
"/cfg.json" and "/presets.json" shows me data.

twlare, 2021-01-07:
@Bebick I just downloaded the latest source code and compiled it and now I have the same problem. It worked last week, but now the edit screen is blank. I will investigate further...

twlare, 2021-01-10:
@Bebick I started with a brand new WEMOS D1 Mini, downloaded the latest source code, compiled it, and copied it to the D1 Mini. I connected to the D1 Mini via the WLED AP at 4.3.2.1. I received no filesystem errors and got the same result as you - /edit shows a screen with only a black bar at the top. I configured the D1 Mini for Wi-Fi access to my network, restarted, and connected to it via 192.168.1.99. Now the /edit screen shows the proper two files on the left - "/cfg.json" and "/presets.json". I think you should change the status of this issue to "bug".

BTW, I have an earlier version of code running my house lights and /edit works correctly when accessing the ESP8266 via the WLED AP. I seems that this problem is fairly new.

Bebick, 2021-01-12:
#

Aircoookie, 2021-01-12:
Thank you for finding out this only happens in AP mode!

This has allowed me to find the issue. The edit page (not written by me) depends on the library `ace.js` which it tries to load from `https://cdnjs.cloudflare.com/ajax/libs/ace/1.2.6/ace.js`. Of course this is not possible without internet access (which there is not in AP mode). I don't know why it even worked before, maybe the library was cached by the browser. Ace is responsible for the editor and syntax highlighting.
I would believe there are two potential solutions to this. The first would be to include ace into the files servable by the ESP. This would make all binaries 80kB bigger though, quite a significant amount. The second would be to add a fall-back to a simple textarea if the library can't be loaded.

twlare, 2021-01-12:
Maybe make the inclusion of the library a compile time option for those who wish to run in AP mode.

Or allow it to retrieved from the built-in filesystem. For example: start with fresh WLED, set up Wi-Fi credentials, download ace.js into littleFS, remove Wi-Fi credentials. When WLED serves up /edit page, it tries to load ace.js from littleFS. If that fails, load from Internet (or vice versa). The download of ace.js could be a manual procedure. This is somewhat complicated, but how many people who run in standalone AP mode are editing the presets.json file (except me)?

Aircoookie, 2021-01-12:
A compile time option indeed doesn't sound bad!

Retrieving it from FS should already work (although I haven't tested it). The built-in DNS server in AP mode re-routes the request to `4.3.2.1/ace.js` where it of course 404's. If the file is present in FS, it should be served :)

twlare, 2021-01-13:
I confirmed two workarounds for using the file editor in AP mode:
1. Visit `https://cdnjs.cloudflare.com/ajax/libs/ace/1.2.6/ace.js` in the Chrome browser. Visit WLED AP at 4.3.2.1/edit - the file editor works correctly. The Chrome browser has cached the ace.js file.
2. Download `https://cdnjs.cloudflare.com/ajax/libs/ace/1.2.6/ace.js` to your computer. Using workaround 1,  connect to WLED AP at 4.3.2.1/edit and upload ace.js to WLED filesystem. Clear the Chrome browser cache and reboot WLED - the file editor works correctly.

ryansabin, 2021-01-14:
I was able to resolve this issue with the following changes.

1. Download ace.js from https://cdnjs.cloudflare.com/ajax/libs/ace/1.2.6/ace.js and place in wled00/data/ace.js
2. Modify wled00/tools/cdata.json add the following right under the favicon block
    {
      file: "ace.js",
      name: "ace",
      method: "binary",
    },
3. Modify wled00/wled_server.cpp add the following right under the favicon function
  server.on("/ace.js", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/html", ace, 354518);
  });
4. You must run the command "npm run build" as explained on the wiki page https://github.com/Aircoookie/WLED/wiki/Add-own-functionality in the "Changing Web UI" section
5. Build and upload your code

embedded-creations, 2021-08-05:
I'm evaluating lorol's fork of ESPAsyncWebServer which is getting updates as opposed to the stale me-no-dev repo.  One of the features is a standalone SPIFFSEditor option:

https://github.com/lorol/ESPAsyncWebServer/commit/406f21c5ddd39a112c5bb90bf95580c39733bde7

blazoncek, 2021-08-23:
What could be done when preparing binaries for flashing is adding a copy of ace.js to a file system.
And while we'd do this we could move all html,css and js files to FS to reduce flash memory use.
Of course this could then mean that a user could (accidentally) delete those files.

embedded-creations, 2021-08-23:
> Of course this could then mean that a user could (accidentally) delete those files.

The `.exclude.files` list feature could be added back so the files could be hidden by default and hopefully are much less likely to be deleted.

https://github.com/Aircoookie/ESPAsyncWebServer/commit/5c274248566a34f9684d85c275e320a5f461d152

embedded-creations, 2021-10-30:
There's a couple SPIFFSEditor issues already, rather than make a new one I'll just add to this one, though there's more being discussed than just the AP mode.

I've made some improvements on top of @lorol's fork of ESPAsyncWebServer, the biggest being support for browsing multiple filesystems using SPIFFSEditor.  I also improved the SPIFFSEditor "list" request so that it doesn't cause a watchdog reboot if listing a long directory takes too long, and it will also work if the directory listing is so long it won't fit into the response buffer.

The AsyncChunkedStreamResponse I made for "list", and the AsyncStreamRepeaterResponse I added but haven't used yet in an example could be useful for WLED in general.  AsyncChunkedStreamResponse lets you use chunked responses as easy as printing to a Stream.  AsyncStreamRepeaterResponse lets you print something that may or may not be larger than the response buffer, without needing to break it into chunks.  You could have a AsyncStreamRepeaterResponse callback that's basically just `serializeJson(doc, printPtr)` where doc is huge, and it will call serializeJson() repeatedly, using the section of the printed JSON that's relevant for the current chunk, and throw the rest out.  This would make the callback coding very simple, but would waste CPU cycles serializing the sections that aren't necessary for the current chunk.  I'm happy to elaborate more and add an example, maybe in a different issue.

https://github.com/embedded-creations/ESPAsyncWebServer/tree/chunkedstream

To compile and run the ESP_AsyncFileBrowser in my fork:

- comment out EDFS in SpiFFSEditor.cpp:  `//#define EDFS`
- add your wifi credentials in ESP_AsyncFileBrowser.ino
- uncomment `#define USE_LittleFS`
- optionally uncomment `USE_SD` and define your SD pins
- on first connect use "admin:admin" user/pass

embedded-creations, 2021-10-30:
Just in case it helps, here's my completed notes from months ago when I was evaluating @lorol's fork:

```
- Issues
	- Creating a LittleFS filename without a suffix or trailing slash makes it think it's a directory though it acts like a file, at least briefly (editor opens)
		- See 3 JUL 2020 commit:
		- "For a compatibility, a folder alone cannot be purposely created. A file will be created instead.
		- Note, a confuse may occur: /mystuff could be a folder or a file, both will look same way on SPIFFSEditor web file tree."
	- No highlighting which file is open in editor
		- Actually shown next to the save button
	- No error when trying to delete non-empty folder, but doesn't work
	- Deleting last file from folder deletes folder (expected with LittleFS?)
	- Can't resize sidebar to fit long filenames
		- make hover show full name?
	- Crashing if files (or filesystem?) containing Ace.js, etc isn't loaded
		- Doesn't crash, just loads a boring incomplete page if `EDFS` isn't defined, and Edit is still in filesystem.  Crashes if EDFS defined
	- Can't open uncommon files
		- e.g. `/.exclude.files` or `favicon.ico`
		- Workaround to type filename, press "create", file opens up
	- SD Issues
		- Creating empty folder returns "error 500", but does create the folder
- Changes from master
	- "A (very ugly) workaround at server-side SPIFFSEditor.cpp (and edit.htm source) for old iOS 9 Safari browsers to save current file through ACE editor w/o file upload allowed natively by the browser. Side effect: the line endings are always Win style after saving due to application/x-www-form-urlencoded enforcement of FormData JS"
		- Is that side effect just for iOS 9?
		- Is this what the "Alt." button does next to save?
	- Ace was updated
		- There's frequent Ace and other js.gz files committed, but at least one diff showed identical binaries after compare
	- memory optimizations
	- allow to "download" with more file types by right-click on file tree
	- edit.htm can be binary-embedded in a firmware or a in gzipped file on fs (saves 4k of program storage, RAM usage is identical)
		- To embed: Comment `#define EDFS` in SPIFFSEditor.cpp and run do_emb.bat
		- On fs: Enable `#define EDFS` in SPIFFSEditor.cpp and run do_ed_fs.bat
	- Added ext-searchbox (Ctrl+F) search/replace functionality and -workers (live syntax checkers) to the ace editor - a standalone hack.
	- SmartSwitch changes that probably didn't make it into main example
		- fixed index.html to be minifyable
		- common authentication for all pages
```

twintersx, 2024-05-27:
Just following up on this issue: has there been any recent development in a functional /edit page within AP mode? It seems @ryansabin method works, but ace.json completely fills up the flash. For my use case, I do not want my home WIFI credentials stored in a product someone else has purchased from me. An Internet-less /edit page would be a phenomenal improvement in the consumer product arena of WLED.

blazoncek, 2024-05-27:
No. It is a component of AsyncWebServer that is fetched from CDN as it is quite large and would consume too much flash memory and make binary file that much larger.

zephyrr, 2024-10-23:
I want to note that this is not just about AP mode; it's also a problem in station mode if the AP/Router is not internet connected.  I use a separate AP/router for displays, for security and non-interference between lighting controls and normal WiFi internet access.  

I cannot use /edit for WLED nodes on the lighting network.  I can:
(1) bring the display controller inside, near my computer
(2) Long Press to get into AP mode
(3) use my computer browser to access the node in AP mode, fiddling until I can get a cached ace.js to work
(4) make the changes (eg: uploading a new ledmapX.json file)
(5) reboot the controller and take it back outside to test on the display
(6) rinse and repeat  (If I upload ace.js, then step 3 is easier thereafter; trying to get the browser to use a cached copy is fiddly)

It would be really great if WLED would automatically use a local file ace.js if it exists, whether in AP mode or not; and only fall back on assuming it's connected to an internet connected router to fetch from cloudflare if there is no local file.

If there is an easier way to do this, I'm all ears.

blazoncek, 2024-10-24:
Ace.js is a large file that will consume too much flash space which is already at premium.

Aircoookie, 2024-10-24:
A majority of ace.js is for formatting which is not really required. I wanted to add a simpler local version with no dependencies that just lists the files and displays the content in a regular text area for the longest time. Maybe I will finally get to it in the next 6 months.

willmmiles, 2024-11-14:
> Just ran into this... no way to include ace.js into the FS while flashing? Sure space is at a premium in the actual code/app partitions, but I doubt many people go anywhere near close to filling up the 1MB FS...

I've looked in to this possibility for the other static web content; I thought maybe it could save some code partition space.  It's possible to build a starting filesystem when doing an initial flash via serial, but after that, extreme care must be taken to disambiguate "initial" flashes vs "update" flashes; and none of the standard OTA updaters have filesystem drivers.  I think it might be *technically* possible to implement a workaround: we'd need to develop a custom OTA uploader that accepts a file with the FS content prepended, so it can be deployed before the code update is handed over to the boot loader for flashing in place; and a whole whack of custom serial upload code that uses a custom "flasher" program that has the filesystem drivers included and accepts new commands to safely write file updates without destroying any other local files and configuration.  And of course we'd have to deal with all the challenges of figuring out how to deal with flash failures and rollbacks if the UI got out of step with the code.

The upshot is it's not, strictly speaking, *im*possible -- but I'd estimate probably a month or two of full-time work to produce something that'd be barely functional, fragile to platform software changes, and difficult to port to new chips.

-----

In terms of short-term mitigations, things in easier reach (at least for me) would be:
- Serving a 302 redirect to cloudflare for /ace.js if it isn't in the filesystem
  - Changing the local edit page to always call on /ace.js
  - Having an option for the browser to upload it if the user wants.  Compressed it's about 90k; big enough that it can't be safely deployed to all clients, but not beyond reasonable for most users
- As above, replacing with a stripped down version

blazoncek, 2024-11-14:
> no way to include ace.js into the FS while flashing?

It is possible to include *anything* in the image for FS. But it is not FS where pre-compiled files for web server go. Those are embedded in the firmware.

Preparing FS image has its own caveats. If you want to upload it, dig the information I (and a few others) posted on Discord. Be aware that uploading FS will erase its content. You can upload FS OTA too.

zephyrr, 2024-11-14:
Trying again.  The title is not quite accurate.  It should be "/edit fails when not connected to the internet"
That can happen when in AP mode, or when in station mode connected to an AP without internet connection (eg: dedicated for shows).  The current title only describes part of the issue.

Using AP mode is usually considered a temporary action, but using a separate non-internet router for running a light show is long term for some of us (for security and coverage), and yet cannot currently support /edit mode.

The simple fix I would prefer is:
* If ace.js is in the file system, use it for /edit mode
* Otherwise, try to fetch it from cloudflare

Then once we get ace.js uploaded (via workaround already discussed), a controller will work in AP mode AND when connected to non-internet AP/routers.

The two options could be reversed, checking the file system after fetching fails, but that takes a bit longer.

I thought it better to discuss this here, rather than opening a new issue "/edit fails when connected in station mode to an AP without internet connection".

blazoncek, 2024-11-14:
FYI `/upload` endpoint works with or without internet access and/or in AP mode. You can utilize it as such if needed.