wled/WLED #1596: Cant "un sync" from hue lights

jimmyeao, 2020-12-29:
**Describe the bug**
After clearing the sync settings for Hue (I enabled for tessting) the wled (11.0) still syncs to the hue lights

**To Reproduce**
Steps to reproduce the behavior, if consistently possible
configure a hue sync, then 'clear' the settings

**Expected behavior**
Wled would stop syncing with hue light

**WLED version**
 - Board: NodeMCU V3
 - Version [e.g. 0.10.0, dev200603] 11.0
 - Format [e.g. Binary, self-compiled] binary

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

MarkBethellHome, 2021-01-17:
Hi @jimmyeao 

I beleive this has been true since at least WLED V0.8.5

I have also noticed another bug relating to the “Hue Sync “ wondered if anyone else has seen this:

Setup: 
Philips Ambilight+Hue TV with two hue lights one on each side driven by the TV. I then have two other WLED lights one sync'd to the left Hue lamp and one sync'd to the Hue lamp on the right. All sounds good :) However both WLED lamps follow the right hand Hue light, which also happens to have the lowest Hue ID number.

Just curious to see if anyone else has seen this problem?

Thanks.

jimmyeao, 2021-01-19:
Not tried that one, but the sync issue is a royal pain :/

jimmyeao, 2021-06-02:
any updates on this?

closed 2022-04-16 by stale[bot]