wled/WLED #1597: Support for sk6812 chip

luminxa, 2020-12-30:
Does wled work with sk6812 yet?

imberest, 2020-12-30:
Yes it does and was working long time ago. In LED setup tick the box for RGBW

luminxa, 2020-12-31:
thank you sorry for the trouble

closed 2020-12-31 by luminxa

oktay89, 2020-12-31:
Can you help me on connecting Wemos D1 Mini with sk6812 (70cm stripe)?
I connected 20a 5v power supply:
V+ power supply -> 5V SK6812 (red)
V- power supply -> GND SK6812 (white)
Data SK6812 -> D4 (Wemos)
5V SK6812 -> 5v (Wemos)
GND -> GND (Wemos)

(I also tried to disconnect 5V and GND and just power the Wemos through USB.)
But no LEDs gets light up. 

I can connect to without any problems to Wemos and also checked the RGBW.

What I am doing wrong?

imberest, 2020-12-31:
Hi @oktay89 ,
Which version are tou using, pre compiled or you are compiling your self?

My wild guess is that you need to turn on leds on app. 
Just to make sure, can you upload example from neopixel and confirm your setup is correct?

oktay89, 2020-12-31:
I use the precompiled bin version:
https://github.com/Aircoookie/WLED/releases/download/v0.11.0/WLED_0.11.0_ESP8266.bin
I think I turned them on, also tried to change color/brightness from app, but just nothing happens.

imberest, 2020-12-31:
@oktay89 then it has to be one of the following (might be a lot more causes):

1. Wiring. Your explained wiring is correct, is it excactly same on your LED+Weemos?
2. Faulty LED or Wemos. For example your first LED is dead and it makes whole strip not to work. Might be GPIO2 is dead. Do some testing and see if they are ok.
3. Are you wiring Weemos D4 pin to Data in on LED strip? Its one way communication. Try different library just to prove your setup is working.

oktay89, 2020-12-31:
Ubfortunately still no success, I tried:
- Checked the arrow direction
- Cut the first led out
- Another wemos
All the time just with data pin in d4 of Wemos and usb power.