wled/WLED #1598: Ws2801 Strip funktioniert nicht

zirius99, 2020-12-30:
Hallo,

ich habe eine Problem mit meinem WLED Setup. Ich habe mir einen AZDelivery D1 Mini NodeMcu mit ESP8266-12F gekauft, um meinen WS2801 Strip damit per WLAN anzusteuern. Habe auch erfolgreich WLED auf dem Board installieren können, aber wenn ich meinen Strip anschließe, leuchten einzelne Segmente des Strips auf, gehen wieder aus und leuchten wieder auf, als würde eine Sequenz ablaufen. Wenn ich einen Befehl über das Webinterface senden möchte, blinken alle LEDs ein paar mal schnell auf, aber die Farbe oder Effekt wird nicht übernommen. Geflasht habe ich die folgende Binary: WLED_0.11.1_ESP8266_WS2801.bin
Die LEDs sind an D3 für clock und D4 für data angeschlossen. Weiß jemand woran das Problem liegt und kann mich auf den richtigen Weg führen?

Viele Grüße
Z.

gsieben, 2020-12-31:
Sehr oft sind schlecht geglättete Netzteile das Problem. Sie stören die Kommunkation. Darum empfehle ich oft einen Kondensator zur Glättung zu verwenden. Ausserdem schützt der Kondensator die LEDs vor Überpannung. Versuche es mal mit mein nur einem Teil des Streifens und versorge den Stripe über den den ESP. Den Streifen vor der Kopplung sollte der ESP von der Belastung her packen. Versorgen den ESP über ein USB Netzteil und schau, ob es dann funktioniert. Am besten stellst Du vorher die Helligkeit auf 128 oder kleiner. Gruß Gabriel

funk01, 2021-01-08:
Hi,
habe das gleiche Board. WLED läuft. Kann mit "Power on/off" LED am Board an und ausmachen. Aber: Stip mit einer oder 10 LEDs bleibt dunkel (D4). Eine Idee? Habe auch schon anderen Strip versucht (WS2812B). Verwendet habe ich WLED_0.9.1_ESP8266.bin

Danke für jeden Tipp.

closed 2021-06-18 by stale[bot]