wled/WLED #1599: WLED on Wemos D1 mini crash

DarthWeber, 2020-12-30:
Dear community,

my wled is running on a wemos d1 mini, 360 LEDs setup as 6 segments. The board is connected to a 5V/10A power adapter, a relay switches on the strip as needed with the build in relay mod.

Everything works perfect until ....

some days or even weeks the board is still connectable with wifi, but not effect seems to work. Only a few leds on the frist segment are on with random colors, but i can use the soft switch in/off with the app. Even no reset helps to get the strips working, also a update with the same or new firmware doesnt't help. 

I need to hard switch of the power consumption of ther wemos, after that everything is working again for days or weeks....

Any ideas what could be wrong ? 

thanks,
tom

gsieben, 2020-12-30:
Hi Tom. That sounds more like an electrical problem than a software problem to me. Have you tried to configure less LEDs in the software. Set the LED count to 30 and see if it works. I guess it will not work. Then take a short stripe and try it with that. Did you put a resistor between the EPS and the stripe? Did you use capacitors to protect the LEDs? 

Sometimes it can also be the power supply. The EPS can with bad voltage. But not the LEDs which communicate via GND. Try it strongly with a capacitor to smooth out unevenness in the power supply. 
Cheer Gabriel

blazoncek, 2020-12-31:
@gsieben it is (or I should say it most likely is) a SW problem. With many segments 8266 runs out of memory after some time. I've implemented a workaround in my fork in custom-buid branch.
On the other hand @DarthWeber, 10A is only enough for about 150 LEDs, not 360.

DarthWeber, 2021-01-02:
Thanks for your feedbacks.
@gsieben resistor yes, cpacitor no (since it should be put close to the strip, but thats not psossible anymore)
@blazoncek if this is a SW problem, why does a reset (software boot ) not help ? 

After switching of the power from the wemos it is runnig now with all segments without any issues now since 3 days.

closed 2021-06-16 by stale[bot]