wled/WLED #1600: Led stripe does not go on after being switched off for a few hours

bert10000, 2020-12-30:
Hello, I have Wled 0.11.1 on a Nodemcu with an external power supply, it works fine during operation. But if I switch off the Led Stripe for a longer time, I can no longer switch it on.
The web interface is accessible and the blue LED turns on when I press the power button, but the LED strip stays off.
If I do a reboot via the web interface or if I briefly power off the Nodemcu, everything works again until I switch it off again for a long time. Does anyone have a solution for the problem?

gsieben, 2020-12-31:
This is an electronic problem and not a bug. Do you use a relay or mosfet to switch off the stripe completely? Have you tried disconnecting the stripe from the power supply and reconnecting it without restarting the EPS? Do you protect your stripe with capacitors? I rather suspect that your stripes don't want something there.

bert10000, 2021-01-02:
i Think it was a Problem with the iobroker Wled adapter, i stopped it and now it works.

medjaiiii, 2021-03-18:
The same problem, I also use iobroker. The device itself is ESP01. How did you decide?

bert10000, 2021-03-18:
> The same problem, I also use iobroker. The device itself is ESP01. How did you decide?

I used the wrong data point in iobroker

medjaiiii, 2021-03-18:
Excuse me, can I go into a little more detail? Not quite clear =(
I think it will be useful for others.

bert10000, 2021-03-19:
> > > The same problem, I also use iobroker. The device itself is ESP01. How did you decide?
> > 
> > 
> > I used the wrong data point in iobroker
> 
> Excuse me, can I go into a little more detail? Not quite clear =(
> I think it will be useful for others.

First i use the "on" data point in the "seg" folder that was wrong.

medjaiiii, 2021-03-19:
> First i use the "on" data point in the "seg" folder that was wrong.

That's right, thanks, I was. it looks like I made a mistake when I changed it in the script.

closed 2021-07-30 by stale[bot]