wled/WLED #1602: Если не стартует ESP32 с прошивкой WLED \ If ESP32 does not start with WLED firmware

DivanX10, 2020-12-30:
Если залили прошивку WLED на плату ESP32 и через USB она работает и можете подключиться к плате, но стоит ее подключить к блоку питания, запитать плату через контакты +5v и GND и плата не работает. Как это исправить? Делается следующим образом. Подключите к контактам как на картинке и плата заработает.

If you have filled the firmware WLED on the ESP32 board and it works via USB and you can connect to the board, but it is necessary to connect it to the power supply, power the board through the +5v and GND contacts and the board does not work. How do I fix this? It is done as follows. Connect to the contacts as shown in the picture and the board will work.

[image]

gsieben, 2020-12-31:
I don't know which ESP32 you have exactly. The 5 V pin is only intended as an output. The ESP32 have a special VIN port for the power supply. Your GND is in a strange place. Is it really there? Happy new year. Gabriel

[image]

DivanX10, 2020-12-31:
У меня две платы ESP32 у которых подключение происходит как на этой картинке и все работает ок. Когда подключил третью плату ESP32, то был удивлен, что она не стартует при таком подключении и стал гуглить. Оказывается, бывают такие платы, что не могут стартовать таким методом подключения. Я нашел для себя рабочий способ и поделился. Возможно кто-то столкнется с подобной проблемой и вот тут уже есть ответ.

I have two ESP32 boards that connect as in this picture and everything works ok. When I connected the third ESP32 board, I was surprised that it does not start with such a connection and began to Google. It turns out that there are such boards that can not start with this connection method. I found a working way for myself and shared it. Perhaps someone will face a similar problem and here there is already an answer.

[image]

EugeneKorneev, 2021-01-04:
Достаточно изучить документацию к плате и общие рекомендации по ее использованию **с любой прошивкой**!
Это не имеет ничего общего с проектом WLED.

Enough you need study the documentation for the board and general recommendations for using it **with any firmware!**
This has little to do with the WLED project.

DivanX10, 2021-01-04:
Верно. Но так как использую данную прошивку, то оставил информацию тут. 

Right. But since I use this firmware, I left the information here.

DivanX10, 2021-01-20:
Для понимания и для наглядности показываю две модели ESP32: ESP32 NodeMCU и ESP32 DevKit V1. Подключение у них разное. 
For understanding and clarity, I show two models of ECO 32: ECO 32 NodeMCU and ESP 32 DevKit V1. Their connection is different.

Одна плата ESP32 NodeMCU подключается таким образом
One ESP32 NodeMCU card is connected in this way
[image]

Вторая плата ESP32 DevKit V1 подключается таким образом
The second ESP32 DevKit V1 board is connected in this way
[image]

[image]

EugeneKorneev, 2021-01-24:
та ладно посмотри на ногу возле 5V !!! так же подключается как и devkit !
а то что ground  можно любую (она общая)
okay look at the leg near 5V !!! connects as well as devkit!
but the fact that ground can be any (it is common)

DivanX10, 2021-01-24:
Тогда почему у меня не стартует? Я и так и сяк пробовал. 
Then why doesn't it start for me? I've tried it this way and that.

ESP32 NodeMCU  если я подключу вот так, то WLED стартует. Не так как у devkit
ESP 32 NodeMCU if I connect like this, then WLED starts. Not like devkit
[image]

ESP32 NodeMCU  если я подключу вот так, то WLED не стартует. Не так как у devkit
ESP 32 NodeMCU if I connect it like this, WLED does not start. Not like devkit
[image]

EugeneKorneev, 2021-01-24:
PS у тебя маркировка китайская!!!

El-Mare, 2021-01-25:
@DivanX10 Other than on ESP-32, on ESP-32s the pin next to 5V is NOT ground (GND), but flash command (CMD/CND -> look closely) and GPIO11. This is why it will not work ;)
[https://images.app.goo.gl/jctFmGPSafvTTKq18](url)

closed 2021-06-02 by DivanX10