wled/WLED #1604: Wemos D1 Mini and NodeMCU not booting

meltedhaggis, 2020-12-31:
I've had a good look through here and tried all the solutions but haven't had any luck, what I have is about 90 leds in a strip WS2811 12v, I'm using a single 12v psu for the job, into a regulator that's giving me the 5v for the D1/MCU, connected up D4 to the data pins with a 470ohm resistor and the strip to 12V, made sure its a common ground all around, it didn't work I thought it maybe because I was only able to buy D1 clones to I bought the NodeMCUS and they behave the same, since then I've tried using a level shifter on both and a 4.7kohm resistor over d4/3.3v

when the data line is connected to the level shifter wled doesn't boot I don't get any access point but if I power down and remove the data line it boots up no problem, I'm guessing D4 is too low but i've no idea how to sort that, id really appreciate any advise.

gsieben, 2020-12-31:
If it does not boot, something is wrong with your circuitry. Be careful, because you can destroy your hardware. Why don't you draw your circuit? Then I can have a look at it. By the way, it doesn't matter that the ESP is a clone. The hardware behind the ESP chips is trivial.

meltedhaggis, 2020-12-31:
Sorry maybe I didnt explain properly It boots fine when I disconnect the data wire, but if I have the data wire connected and power it up the blue light comes on and the led strip lights up but I get no wled ap everything appears to be fine I just don't get any WiFi, if I power it down and disconnect d4 it boots up fine with WiFi, I may have messed up wiring the logic shifter I've never done it before I really appreciate your assistance, I'll find a pad and note down what I've done and come back shortly with it, that's great to know about the clones I was concerned they where so cheap

meltedhaggis, 2020-12-31:
[image]

I've never drawn a diagram like this before I hope it's good enough to understand, I read pin d4 has to be high for the esp to boot and possibly the LEDs are bringing it down but I attached the 4.7k ohm in the diagram and it didn't help

meltedhaggis, 2020-12-31:
[image]
I'll stick this up aswell, I don't do little projects like this often but I solder for my job, I want to make a few of these so I tried to make it half decent, I'm confident my soldering a correct and everything's going where it should, I think it's just something I don't understand yet, I don't get the high and low pin to boot thing but I think it's my problem

Aircoookie, 2020-12-31:
Hi!
You should be able to use a resistor as low as 330ohms (recommended 470ohm) between D4 and 3.3V :)
The transistor-style type of levelshifter you are using could also be too slow to drive the LEDs reliably. I mostly recommend the sn74ahct125n chip. In many cases your type of levelshifter might work though, so don't rip it out immediately.

I hope you'll be able to get your board to work and happy new year :)

meltedhaggis, 2020-12-31:
Hey Aircookie, I really appreciate the reply, I literally just read your post about how swamped you are so I really do appreciate it, perfect I'm away out now but I'll try swapping out the 4.7k for a 470ohm in the morning, if I have no joy with the controls I'll get the IC you recommended.

Thanks again I hope you have a good one, happy new year bud

meltedhaggis, 2021-01-01:
So I swapped out the 4.7kohm for a 470ohm and now the esp boots! I get an access point and the led strip lights up white however I can't control it all, I tried all the settings in led preferences none seemed to work, could 470ohm possibly be too small and allowing too much from 3.3v through? Is there a calculation that cane be done to figure out the right resistor if that's the case?

Hope everyone had a good one last night

meltedhaggis, 2021-01-01:
So I've just solved my issue, would seem I had two problems incase anyone else has the same issue.

There was something wrong with a section of my led strip, I'm guessing I damaged it when I wired the D1 up without the level shifter originally, because it was damaged when I connected up d4 the esp didn't boot.

My second issue that was making it hard to diagnose was I didn't need the 470ohm resistor on d4, I removed the resistor and the 30 LEDs went orange beautifully

Blightbuster, 2021-03-17:
Im having a simular problem. My D1 Mini also does not boot when data is connected to D4. It works find when disconnection D4 and then reconnecting once its booted.

I have no resistors or level shifters in my circuit at all.

I suspect it has to do with the D4 being pulled low on boot which is supposed to be high ( [reference](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/) ).

I will try with some of the WLED_0.x.x_ESP8266_ledpinY.bin releases and give a quick update if any of those pins work.

[image]

Blightbuster, 2021-03-17:
Using the `WLED_0.x.x_ESP8266_ledpin4.bin` and connecting **DATA** to **GPIO4** ( labeled as **D2** on my D1 Mini ), works flawless. 👍

closed 2021-07-22 by stale[bot]