wled/WLED #1606: TM1914 Chip Support

luminxa, 2020-12-31:
I thought I ordered sk6812 but turns out its tm1914 chip.. any support for this?

Def3nder, 2021-01-05:
Hi @luminxa, 
I think you will be out-of-luck with this chipset as there is no support for the TM1914 in the FastLED library...
...no does the NeoPixel Brightness Bus library support this.

(the protocol seems to be "inverted" and not that easy to handle).

So you better order a SK6812 ...

oraclesod, 2021-05-26:
Comparing the datasheet from the TM1814 and TM1914, they are very similar, both are inverted too... Different OUT PWM output frequencies and Propagation delay times...

Would anyone be able to make the changes to allow the TM1914 ? I've attached its datasheet, but im not very good at coding so not sure ill be able to do this myself :(

[TM1914A_V1.1_EN.pdf](https://github.com/Aircoookie/WLED/files/6544758/TM1914A_V1.1_EN.pdf)

PR #3913 (2024-05-10): Tm1914 chip support
Closes #1606
closed 2024-10-27 by blazoncek