wled/WLED #1607: 2 MB flash drive vor 4MB devices

gsieben, 2020-12-31:
The D1 Mini has 4 MB of flash memory. Why waste this space? At the moment 1 MB is not used.

In my opinion the best setting is: 2MB flashdrive and 1MB OTA memory. 

For this the buildflag: `-Wl,-Teagle.flash.4m2m.ld` would have to be included. 

File: `platformio.ini`
 
[env:d1_mini]
build_flags = ${common.build_flags_esp8266} 
    -Wl,-Teagle.flash.4m2m.ld ; 2m SPIFFS & 1M OTA

Same for:
[env:d1_mini_debug]
[env:d1_mini_ota]

Generally this applies to all devices with 4 MB or equivalent 8MB

[image]

Aircoookie, 2020-12-31:
Hi!
Thank you for the suggestion! I believe I will add a custom environment for it 👍 

Actually I would have loved to use this layout by default in 0.11, but when I OTA upgraded my 0.10.2 installation to 0.11.0 the EEPROM settings were lost. That's why I sticked to 4m1m. Now a quick look into the ldscript seems to indicate that the EEPROM page is behind the FS at the same memory offset, so it _should_ have retained the EEPROM and I don't know why it didn't for me at that time.

Now that we are using the filesystem already, resizing it would definitely result in a format as you said. Right now I don't really see the need to make the FS larger as it is unlikely that preset and config file would grow beyond about 500kB. When the need arises (perhaps new features like playing .fseq files from FS or displaying images on a matrix) we'd need to migrate, but that would require you to manually backup and restore every single existing installation.

I wish there was a way to utilize 2MB FS automatically only on new installations. Maybe I should add separate binaries for that. Then again I'd risk people flashing the wrong one and losing their presets 😬 

Happy new year :)

gsieben, 2020-12-31:
Oh yes, you're right. That is the problem. The settings are lost. I saw a thread about not enough flash memory. At least that's what it said. I immediately thought of my matrix clock, which I have configured to 2MB. I load all HTML requests from the file system. Makes the whole thing very flexible. HTML changes don't have to be compiled but only uploaded. Unless you need more functionality from the server. Json is a wonderful interface, which you already use. From the response time I don't see any disadvantage. At least not felt. Whereas my clock has only a framerate of about 5 FPS. I don't need more than that. 

Unfortunately it is not possible to understand that the files are not there anymore. The interesting thing is that you can overwrite the current version with the 2MB version. Then the settings are gone. If you install the 1MB version again, they are there again. The settings are obviously not even overwritten. But you have the wonderful /edit. There you can save the settings. Maybe it's not stupid to allow saving the config in the settings anyway. With a few steps and the settings can be reloaded when booting a system without to much efford. I could design something for this. Meanwhile I am familiar enough with the code, thanks to my relay project. Just as a suggestion. But the need for more memory must be real of course...

Happy new year, Gabriel

blazoncek, 2023-05-27:
Shall we close this issue as most (or very large amount) of new installations are done on ESP32 and ESP8266 is slowly becoming a *legacy* device?