wled/WLED #1611: Open Sound Control (OSC) support

dandantsui, 2021-01-01:
**Is your feature request related to a problem? Please describe.**
Trying to get this to talk with MaxMSP but I find myself having to stitch raw UDP signals over to send data to the WLED which is really cumbersome.

**Describe the solution you'd like**
Implement some kind of OSC UDP support -- for example rgb_hex messages would be "/rgb_hex xxxxxx" or presets becomes "/presets xxx"

**Describe alternatives you've considered**
HTTP/JSON requests but they are just as cumbersome when making such big changes.

**Additional context**
It adds on top of UDP communications and opens up the ability for other programs based on music programming to take part.

Thank you for your ideas for making WLED better!

DJBJ0, 2022-06-10:
+1 for this. 

Could also make a simple controller using the many osc programs available over all the platforms such as touchosc, osc pilot, lemur, kiosc, open sound control, and also bridges between different protocols using a program like ola on raspberry pi

softhack007, 2022-06-13:
If i understand correctly, OSC "just" defines a way to transport "stuff", without defining what this "stuff" is. For me this seems to be a similar approach as MQTT. 

To get started, we need to answer some question:
* What "kind of stuff" should be communicated over OSC? If you want OSC in WLED, what kind of services should be offered? What information should be made available?

.

The OSC Homepage https://opensoundcontrol.stanford.edu/index.html has a paper that explains:
> OSC is often referred to as a protocol, but it is only a protocol in the weakest sense in that it defines a message format -it does not define typical features of protocols such as processing semantics (e.g. command-response patterns), error handling or negotiation. It is more accurate to describe OSC as a content format. This means that OSC can be viewed and compared with other formats such as XML [21], WDDX [18] or JSON [4]. An application that uses OSC only guarantees compatibility with OSC  parsers/formatters.

> **Inter-application` protocols are of course possible using OSC as the underlying format, but the syntax and semantics of those interfaces are beyond the  scope of the OSC specification.**

BananaNeil, 2023-05-08:
I would also really love to see this happen. Is there nothing in place for it at the moment?

x2concepts, 2023-08-04:
Anyone made some progress on getting wled mapped in OSC?

8puslab, 2024-03-24:
I'm trying! it's possible to do it but someone (not me I am a newbie  with code) could write a Javascript code that could be used in conjunction with maxurl