wled/WLED #1612: Teensy 4.1 support

Pixel-Therapy, 2021-01-02:
My use case, I'd like to run 16*144 strips in parallel, mostly to use UDP streaming over wired ethernet.
I don't know how difficult it would be to compile a version for Teensy 4, only that [FastLED already has support for it ](https://github.com/FastLED/FastLED/wiki/Parallel-Output#parallel-output-on-the-teensy-4)

DeviousPenguin, 2021-01-03:
Going for dual ESP32 may be worth considering, each ESP32 should be able to to 8 data pins with the multi pin user mod, but wired ethernet is still something that is being worked on I think

Pixel-Therapy, 2021-01-04:
I have considered dual esp's and while that certainly would work and no one would ever notice, it's not very elegant.

AnthonyJQQ, 2021-10-14:
> I have considered dual esp's and while that certainly would work and no one would ever notice, it's not very elegant.

Not very elegant but it works great when using multiple esp32s. I have a matrix with 7420 LEDs controlled between 4 esp32s and the sync function works well to sync the controllers together.

softhack007, 2023-01-06:
Currently there are no plans to add support for Non-ESP boards.

closed 2023-01-06 by softhack007

blaqvoid, 2024-11-01:
> Not very elegant but it works great when using multiple esp32s. I have a matrix with 7420 LEDs controlled between 4 esp32s and the sync function works well to sync the controllers together.

@AnthonyJQQ is that working OK? I'm doing a display with 10k LED's, and wonder if multiple ESP's will sync ok when doing audio reactive 2d matrix effects, or if it will look lousy