wled/WLED #1615: Random Reboots on ESP8266

jhnwmr, 2021-01-03:
Hello,
I am running WLED on an ESP8266. It is a great option to control WS2812 Leds. Unfortunately I still have a problem which I can not localize. No matter what effect I use or what brightness is set, at some point the ESP reboots all by itself. This can be within 30min or within a day. It is completely random and you can not recognize any scheme. The power supply is powerful enough and the voltage does not drop.
Then I used a WS2815 LED strip with 12V power supply for the LEDs and a 5V power supply for the ESP8266. The same case happens that after some random time the strip goes off and and the ESP restarts. On both ESPs I have installed the latest WLED version. The rest works fine. The wifi access point is maximum 5m away.
What else is there to look out for?

maximecb, 2021-01-04:
How confident are you that your 12V power supply is reliable? I've had some issues with my cheap 5V PSU from China being very unstable, fluctuating voltage based on the current demands of the LED strip.

Another issue would be if you don't have a large filter capacitor close to the LED strip. The power demands can cause interference with the ESP8266's onboard voltage regulator. You can try adding a capacitor at the LED strip and at your ESP8266's input, see if it stabilizes thing. Also beware of cheap components from China. There have been fake capacitors :/

nodiaque, 2021-01-14:
I think I have the same problem. I'll have to check mqtt if there's a log on online/offline, but randomly, the strip would flash to the preset 1 and then back to the actuel preset.

I'm unsure it's a reset since it does return to the actuel config, even if not a preset, which shouldn't be possible if i remember correctly (current settings aren't saved on reboot). 

I have 2 wled but since the other one is outside, I'm unsure if it has the same problem. 

Both ws2815, 12v led. 2 differents PSU. Right now, I have a 400W powering a 60 11m, which is more then enough, especially that I'm at 25-30 brightness out of 255.

closed 2021-06-09 by stale[bot]

steppel, 2021-12-21:
Same issue here. 

I have for my cabinets one esp8266 and one wemos d1. Both are powered from the same PSU and only the wemos D1 randomly restarts. It happens what ever is set (effects or solid, different brightness). When it reboots it ends up with the red light. So it didn't keep the last state, looks like a complete reset to initial setup stage except it kept the Wi-Fi config.

blazoncek, 2021-12-21:
Check binaries from @srg74 especially @blazoncek fork. It has some stability improvements as well as experimental features.
Backup your presets and config prior to update.

seifstst, 2022-03-01:
anyone solved that problem ? i have arduino mega with esp8266 inbuilt

seifstst, 2022-03-02:
deletes the ESP versions you have in: C:UsersyouraccountAppDataLocalArduino15packages
download esp8266 version 2.5.2
selected with wemos D1 R1 card
it works very well for me for a long time without interruption