wled/WLED #1616: Effect stops operating after ?random? time period

swampdogmash, 2021-01-03:
**Describe the bug**
Candle effect stops functioning after a random (unconfirmed) amount of time when 40 led strip is split into 2 sections (unconfirmed if it happens in other configurations as yet) and is selected using a favourite. User interface seems to still function and going into the interface and reselecting the favourite starts the effect again. User interface shows the effect on segment 2 as solid. Reselecting the favourite changes the effect on the segment back to candle and all woks again for a period until it stops again

**To Reproduce**
Enable 40 leds, and configure the following favourite on preset position 9.

`{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":20,"grp":1,"spc":0,"on":true,"bri":41,"col":[[8,255,0],[0,0,0],[0,0,0]],"fx":0,"sx":15,"ix":81,"pal":0,"sel":false,"rev":false,"mi":false},{"id":1,"start":20,"stop":40,"grp":1,"spc":0,"on":true,"bri":108,"col":[[255,160,0],[255,224,160],[0,0,0]],"fx":88,"sx":186,"ix":255,"pal":2,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}`

**Expected behavior**
first 20 leds should be solid green - works consistently
remaining 20 leds should be candle effect with 2 shades of orange. This should just continue indefinitely.

**WLED version**
 - Wemos D1 Mini
 - WLED Version 11.1, 11.0
 - Full Binary (not 2mb)

**Additional context**
WLED is running on the Wemos D1 mini, and powered from a 5v 2.0 power supply. A 74hc404 is being used to change the 3.3v D4 output to 5v. LEDs are 12v and are powered independently (common negative) by a 12a 12v power supply. I am going to do some more testing with different combinations (no segments, different effect etc) to see if this occurs with anything else.

Thanks for some awesome software!

swampdogmash, 2021-01-03:
Looking at the preset file (after noticing the number of stop statements in the preset I posted above) I did notice that preset 9 has 10 stop statements and all the others have 11.

`		{
			"stop": 0
		},`

Multiples stops doesn't seem right anyways.. Am i doing something wrong in the GUI to have produced this?

Full preset file here.
[https://pastebin.com/qNkWkRqT](url)

swampdogmash, 2021-01-04:
To add to this - This has happened over periods of 30 mins or so up to say 8 hours and has occurred around 10 times though since i logged this strange behaviour above, it has been consistently running for 18 hours - shall update once it happens again.

swampdogmash, 2021-01-04:
Having left the effect running for 30 hours and it's now done the same thing again. the effect on segment 2 has changed to "solid" and for obvious reasons the colour is now... er.... solid....

Can't think of anything currently that was happening to it to cause this - no logins to the UI, or MDNS requests that wouldn't have happened previously etc. Bit of a strange one tbh.

Def3nder, 2021-01-05:
I can confirm this does happen on ESP32, too (over night the candle effect changed to solid color) - will do further tests.

snickers2k, 2021-02-15:
dont know if this is the same issue. but for me it's changing to greenish color after a few hours :/

swampdogmash, 2021-02-16:
TBF, I've seen it with another effect on a different set-up - though it did also include multiple segments. It's not that it actually freezes, it actually seems to revert to "Solid" as the pattern.

Aircoookie, 2021-02-18:
Hi!
Giving it a try now, but with your effect JSON, Segment 1 (the one with Candle effect) is selected. Are there other WLED instances in your network which could be sending a sync signal to the unit with the issue? You can try selecting Segment 0 only instead (by setting the checkmark left to the segment name) and see whether the problem still occurs.

closed 2021-06-28 by stale[bot]

AndyRPH, 2021-10-03:
So I can reproduce this easily.  Just setup the current .13beta on a SP108E.  Sk6812  strip.   I can  set it to something like candycane effect.  And it'll flip back to default color and solid effect within a couple hours.  Changed it's IP address too so I'm sure no other sessions interacting with it.  Seems effects just won't stick.   

Any log it can create as to what's triggering it to change?

blazoncek, 2021-10-04:
Check the uptime of your wled after the color change.

AndyRPH, 2021-10-04:
Bingo! It's rebooting for some reason.  Hmm.

blazoncek, 2021-10-05:
Serial monitor & debug build will help.