wled/WLED #1619: How to add a button to cycle through Preset Favourites

comtsamsmith, 2021-01-04:
I want to put a flashing row of LEDs to my son's bike.  It would be really nice to set some preset patterns and then wire a button to cycle through the presents (No WiFi on the road).  

Is that possible without me having to change the code?

I would guess it does it on one of the spare inputs but cannot see anything in the documentation.

WoodyLetsCode, 2021-01-08:
WLED does support a button. Connect a normally open pushbutton to GPIO0 (NodeMCU/Wemos pin D3) and ground.
In settings you can configure your button: 
[image]

comtsamsmith, 2021-01-08:
Thanks for that I will try it.

WoodyLetsCode, 2021-01-08:
Alternatively WLED also can create a wifi network so you can connect to WLED also on the road.

comtsamsmith, 2021-01-13:
No sorry, cause I have not had any time to try it.  I gues that will only do the three presets.  But thanks for the tip.  

I did find a in a video at 19 mins in this bloke explain how to cycle through the presents using a button to activate a macro.
https://www.youtube.com/watch?v=6eCE2BpLaUQ&t=922s&ab_channel=DrZzs

Diranar, 2021-05-05:
> No sorry, cause I have not had any time to try it. I gues that will only do the three presets. But thanks for the tip.
>
> I did find a in a video at 19 mins in this bloke explain how to cycle through [...]

This works like a Charm with Version 0.10.2. But will NOT work with all newer Versions, cause now you can only assign 3 different Presets to the Button, but can't change to the next Preset cause this is no longer supported or is broken.

muhackl, 2021-07-12:
I tried a button on pin 0 and GND on a WEMOS Lolin D32 (ESP32) and D3 and GND on a WEMOS D1 Mini (8266) with the latest version of WLED. I added 3 presets and connected them to the 3 button-press-patterns in the settings. But it did not work. The presets did not change after pressing the button. What could have gone wrong or what mistake did I make?

kdegraaf, 2021-07-22:
The following worked for me on v0.12.0 build 2104020:

1. Create preset, uncheck "Overwrite with state", API command "win&P1=1&P2=5PL=~", save to ID 250.  (Replace P1 and P2 with the first and last "real" preset IDs.)
2. Config -> Time & Macros -> "Button short press Preset" -> 250
3. Wire the button per the directions.  (TL;DR: momentary SPST, normally open, across a digital pin and ground, pulled up to 3v3.)

Hopefully, v0.13 will make this easier, but I'm not inclined to mess with the betas.

Good luck!

Mikprice, 2021-07-28:
oh my god thank you SO much @kdegraf !!! This is exactly what i needed, been looking for quite some time...
I use Wled in the speedometer of my car, and needed a way to easily cycle through different colors without always needing to connect my phone first. This works perfect!

dkebler, 2021-08-16:
I'm using .13 beta 2 and this still works.   No documentation on this so no way to know if the devs have made this easier altough this isn't bad

epchris, 2022-02-02:
> 1. win&P1=1&P2=5PL=~

Kevin! I went to college with you :)
This super helpful tip got me on the right path and I wanted to thank you

blazoncek, 2022-02-02:
> Hopefully, v0.13 will make this easier, but I'm not inclined to mess with the betas.

Every software is beta nowadays. If you are wondering, 0.12 has more bugs than 0.13-b6. Nastier too. 😉

blazoncek, 2022-02-02:
For future reference: https://kno.wled.ge/interfaces/http-api/#presets
Closing.

closed 2022-02-02 by blazoncek

loverboyritesh, 2022-03-04:
Thanks @comtsamsmith ... for asking this question, this was something i also needed  and looking for .And 
Thanks @kdegraaf ... For your Awesome trick..