wled/WLED #1622: LED's won't stay off when powered off via the GUI

rich-benson, 2021-01-04:
Hi all. 

I'm new to WLED (and Github / NodeMCU etc in general). I have WLED version 0.11.1 running on a NodeMCU. When the LED's are powered off using the power button within the WLED GUI, or by using a command from Home Assistant, the lights go off as expected. But then after a random amount of time, a couple of LED's will illuminate, usually at the start of the set of LED's, and the colours will be completely random. I thought it may be a bug in Home Assistant, so I removed it, but the problem persists. 

I then thought it may be something to do with how it sleeps, so I turned on "disable wifi sleep" and that made no difference either.

I have 50 12v LED pixels (string, not strip), (ws2811) powered by a 30amp supply. The same supply also powers the NodeMCU via a 12v - 5v convertor.  The router I am using is an ASUS DSL-AC88U. The 2.4ghz network is set to "N" only, and has a separate SSID to the 5ghz network. I set the 2.4 to use all auto settings too, and that made no difference either.

I have played around with my wifi settings etc but I can't get the LED's to stay off without actually killing the power. 

Sometimes, as well as the above, WLED becomes unresponsive, other times, I am able to access the interface and choose an effect etc.

Any ideas? 

Thanks in advance guys&girls, much appreciated.

-Rich

rich-benson, 2021-01-04:
UPDATE:

I know what is causing the problem. I just don't know how it is causing the problem haha. 

In our house we have a 3 speed pump which sends hot water around all the radiators. I decided to switch it to speed 1 from speed 2, and as soon as it changed speeds, the LED's connected to my sleeping WLED system partially came on. I flicked the speed back to 2 and some more LED's came on. How weird is that! I did it again measuring the power through the supply feeding the LED's etc thinking it was spiking or dropping, and it never budged. The voltage was constant. So it must be some kind of interference from the pump affecting the NodeMCU!

sansillusion, 2021-01-04:
LEDs are probably picking up static trough the data wire, is the data wire from esp to LEDs long ? Mabe try a shielded cable ? Or like many others you could use a relay to shut down power to the LEDs when turned off. Make sure your ground is connected from leds to esp too.

rich-benson, 2021-01-04:
> LEDs are probably picking up static trough the data wire, is the data wire from esp to LEDs long ? Mabe try a shielded cable ? Or like many others you could use a relay to shut down power to the LEDs when turned off. Make sure your ground is connected from leds to esp too.

Thanks for your reply, much appreciated. The data wire is very short (as I don’t have sacrificial pixel etc to boost signal). I did change the wire but the problem is the same. 
I have moved my LEDs away from near the heating pump where I was testing, and so far they seem ok. 
So it’s definitely intereference, and hopefully it won’t be a problem when I eventually buy more LEDS and stick them on the house 🤣, 

Thanks again for your help.

closed 2021-01-04 by rich-benson