wled/WLED #1623: Unintentional rainbow, unexpected colors

maximecb, 2021-01-04:
I would appreciate your input as to what might be causing this issue. I have a strip of 85 LED, which as far as I know is a WS2812B, 5V strip. I connected it to GPIO2 of a WeMos D1 Mini using a level shifter using [the second circuit from this page](https://www.hobbytronics.co.uk/mosfet-voltage-level-converter).

I was able to flash the default WLED ESP8266 image and I can access the module's web interface. The UI is working correctly. However, any color that I try to set, I just get a weird rainbow displayed on the strip.

Here's what makes this extra strange:
- I wrote my own firmware using the Arduino IDE and *I can get this strip to work fine and display the colors I want*.
- I have a second LED strip of a different make, with a higher density of LEDs, and it works as expected with the WLED firmware

Extra notes:
- Both strips work normally with my custom firmware and the same circuit
- I've tried turning the RGBW setting on and off and it unfortunately doesn't fix the problem.
- The rainbow is static (not moving or flickering) after the transition period

I would very much appreciate feedback from someone who has more experience with WLED. Are some LED strips just dodgy? Are there common reasons for them not behaving the same?

Def3nder, 2021-01-04:
Hi @maximecb ,

weired rainbow-like colors can be caused from timing issues.

Did you try to strip the level-shifter away (don't use it) ?
A WeMos D1 mini can drive the strip with it's 3,3 V if the first LED of the strip isn't that far away (e.g. below 1m).

It could be the case that the level-shifter circuit is too slow, causing weired problems.
If you need to use a levelshifter (because of a long wire from the WeMos to the strip) then I would recommend to use an 
SN74AHCT125N. 

I hope this does help.

(I personally did not use any level-shifter for WLED at all - and I do have a douzend of them in my house).

sansillusion, 2021-01-04:
This sounds like your strip might be rgbw instead of plain rgb. Try setting
rgbw in led settings.

maximecb, 2021-01-04:
@Def3nder it seems I need the level shifter with this board for some reason, it doesn't work otherwise :woman_shrugging:

@sansillusion I mentioned trying that in my post. 

I did some more investigation and I can produce glitches on both strips, the problem is intermittent, so I think that both of my LED strips are good.

The 5V power supply that I was using is dodgy, it did not maintain a stable 5V at all (bit by the Chinese fakes), so I replaced it. That makes the strips much more stable, but it unfortunately doesn't eliminate the problem completely. I also added a 1000uF capacitor directly at the strip. That makes the colors more stable too.

One issue I noticed is that, I soldered the 470Ohm resistor inline in the data wire going to the LED strip. If I wrap my fingers around the resistor, the LED strip goes crazy, data transmission is perturbed. Maybe because I'm inducing capacitance. That doesn't happen if I wrap my fingers elsewhere around the strip. This may be the last problem that I need to solve to get this working.

maximecb, 2021-01-05:
I was able to get it working smoothly by removing both the level shifter and the 470 Ohm resistor in series with the data pin.

closed 2021-01-05 by maximecb