wled/WLED #1624: Easy UX: Allow a time controlled preset to specify ON or OFF in addition to a preset number

larrywal, 2021-01-05:
A ton of people (me included) just want to do a simple "turn my lights on at XYZ time" and "turn my lights off at ABC time".

What exists is great if you understand how presets work, but even if you do, it would be great to support a lower friction way to create a time controlled "turn my lights off" or "turn my lights on and have them start on whatever the default is" (i.e. what would happen if I physically pressed the button).

One way you could do this is to allow the user to type in ON or OFF instead of a preset number.  Then, for example, I wouldn't have to create a dummy preset with T=0 (which no one understands how to do anyway).  You just need a small note in the UX under the preset grid saying to use ON or OFF instead of preset number where desired.

Aircoookie, 2021-01-08:
I agree! I think I will include this with a future rework of the settings pages!
Even if you know how API presets work, you could save the clutter of having to set up on and off presets just to make a simple schedule. And if you don't, it will make the timed macro feature more accessible :)

balucanb, 2021-06-20:
Bump. Love this idea- Hope we see it soon.

larrywal, 2021-06-20:
:) I'm still looking forward to it too. Btw instead of "turn on to default preset" I think I'd want it to turn on to whatever I last had it at. Maybe do that if "startup preset" is blank.

balucanb, 2021-06-20:
I just need anything that does not require me to learn to write JSON. I have tried to learn, I checked out many "absolute beginner" tutorials....The people who wrote those obviously have a different definition than me. I need kindergarten or brain dead tutorials myself..still might be a reach.

blazoncek, 2023-01-25:
Just create two presets:
1. "On" with API `{"on":true}`
2. "Off" with API `{"on":false}`

And use those in Timed presets.

closed 2023-01-25 by blazoncek